uzh-rpg/agile_autonomy #10: Open3d not found while building catkin workspace

den250400, 2021-10-17:
**Environment:** Ubuntu 20.04, ROS Noetic

I have installed ROS according to instructions from the official website (http://wiki.ros.org/noetic/Installation/Ubuntu) and followed your instruction line-by-line. However, the following error occurred when I tried `catkin build` step.

Would appreciate any help.

`---------------------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/denis/agile_autonomy_ws/catkin_aa
---------------------------------------------------------------------------------------------
Build Space:        [exists] /home/denis/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /home/denis/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /home/denis/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /home/denis/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /home/denis/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
---------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------------------------------
[build] Found '71' packages in 0.0 seconds.                                                                                                                                                               
[build] Package table is up to date.                                                                                                                                                                      
Starting  >>> catkin_boost_python_buildtool                                                                                                                                                               
Starting  >>> catkin_simple                                                                                                                                                                               
Starting  >>> decomp_util                                                                                                                                                                                 
Starting  >>> mav_msgs                                                                                                                                                                                    
Starting  >>> mav_state_machine_msgs                                                                                                                                                                      
Starting  >>> mav_system_msgs                                                                                                                                                                             
Starting  >>> motion_primitive_library                                                                                                                                                                    
Starting  >>> open3d_conversions                                                                                                                                                                          
Finished  <<< catkin_boost_python_buildtool                     [ 0.1 seconds ]                                                                                                                           
Finished  <<< mav_state_machine_msgs                            [ 0.4 seconds ]                                                                                                                           
Starting  >>> rotors_comm                                                                                                                                                                                 
Finished  <<< catkin_simple                                     [ 0.2 seconds ]                                                                                                                           
Finished  <<< mav_system_msgs                                   [ 0.5 seconds ]                                                                                                                           
Finished  <<< mav_msgs                                          [ 0.7 seconds ]                                                                                                                           
Starting  >>> assimp_catkin                                                                                                                                                                               
Starting  >>> catkin_boost_python_buildtool_test                                                                                                                                                          
__________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << open3d_conversions:make /home/denis/agile_autonomy_ws/catkin_aa/logs/open3d_conversions/build.make.006.log                                                                                  
In file included from /home/denis/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/src/open3d_conversions.cpp:15:0:
/home/denis/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19:10: fatal error: Open3D/Open3D.h: No such file or directory
 #include <Open3D/Open3D.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/open3d_conversions.dir/build.make:63: CMakeFiles/open3d_conversions.dir/src/open3d_conversions.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:305: CMakeFiles/open3d_conversions.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/denis/agile_autonomy_ws/catkin_aa/build/open3d_conversions; catkin build --get-env open3d_conversions | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

..........................................................................................................................................................................................................
Failed     << open3d_conversions:make                           [ Exited with code 2 ]                                                                                                                    
Failed    <<< open3d_conversions                                [ 0.5 seconds ]                                                                                                                           
`

MaxTianyy, 2021-11-02:
> > I think that for your version of open3d you need to rename the include file of open3d to `#include "open3d/Open3D.h" `. An example is here: https://github.com/isl-org/open3d-cmake-find-package/blob/master/Draw.cpp#L29.
> > Hope it helps!
> 
> I have been stuck at the same error for 2 full days now but changing to #include "open3d/Open3D.h" has not solved the issue for me. I have tried
> 
> 1. downloading all the dependencies listed in Open3D page,
> 2. changing Open3D installation directories
> 3. renaming #include "open3d/Open3D.h.in" none of this have worked thus far.
> 4. replacing open3d_conversions with the files found here : https://github.com/ros-perception/perception_open3d
> 
> This is the error log I keep getting:
> 
> Errors << open3d_conversions:make /home/csoter01/agile_autonomy_ws/catkin_aa/logs/open3d_conversions/build.make.010.log In file included from /home/csoter01/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/src/open3d_conversions.cpp:15:0: /home/csoter01/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19:10: fatal error: open3d/Open3D.h.in: No such file or directory #include <open3d/Open3D.h.in> ^~~~~~~~~~~~~~~~~~~~ compilation terminated. make[2]: *** [CMakeFiles/open3d_conversions.dir/build.make:63: CMakeFiles/open3d_conversions.dir/src/open3d_conversions.cpp.o] Error 1 make[1]: *** [CMakeFiles/Makefile2:305: CMakeFiles/open3d_conversions.dir/all] Error 2 make: *** [Makefile:141: all] Error 2

I have been stuck at the same error and it's all settled now.
1. please make sure you have installed open3d by building it from source.If so you will see open3d directory in path "/usr/local/bin/cmake/open3d"
2. otherwise,  building open3d from source[http://www.open3d.org/docs/release/compilation.html#compilation](url)
3. renaming #include "open3d/Open3D.h"

den250400, 2021-11-03:
> ```
> $ rviz --v
> $ [ INFO] [1634925235.339356078]: rviz version 1.13.3
> $ [ INFO] [1634925235.339459957]: compiled against Qt version 5.9.5
> $ [ INFO] [1634925235.339489131]: compiled against OGRE version 1.9.0 (Ghadamon)
> ```
> 
> This is a pretty old version, but with minimal changes it could also work with newest versions.

@antonilo @kelia still having problems with rviz, I would appreciate it if you will clarify what do you mean by "minimal changes". Any instructions on how to downgrade rviz to 1.13.3 will be also very helpful.

MaxTianyy, 2021-11-05:
> > ```
> > $ rviz --v
> > $ [ INFO] [1634925235.339356078]: rviz version 1.13.3
> > $ [ INFO] [1634925235.339459957]: compiled against Qt version 5.9.5
> > $ [ INFO] [1634925235.339489131]: compiled against OGRE version 1.9.0 (Ghadamon)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > This is a pretty old version, but with minimal changes it could also work with newest versions.
> 
> @antonilo @kelia still having problems with rviz, I would appreciate it if you will clarify what do you mean by "minimal changes". Any instructions on how to safely downgrade rviz to 1.13.3 will be also very helpful.
> 
> **Update:** I also encountered this error when catkin tried to build traj_sampler
> 
> ```
> Errors     << traj_sampler:make /home/denis/agile_autonomy_ws/catkin_aa/logs/traj_sampler/build.make.002.log                                                                                              
> In file included from /home/denis/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/traj_sampler/src/kdtree.cpp:1:0:
> /home/denis/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/traj_sampler/include/traj_sampler/kdtree.h:5:10: fatal error: Open3D/Geometry/KDTreeFlann.h: No such file or directory
>  #include <Open3D/Geometry/KDTreeFlann.h>
>           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> make[2]: *** [CMakeFiles/kdtree.dir/build.make:63: CMakeFiles/kdtree.dir/src/kdtree.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:1462: CMakeFiles/kdtree.dir/all] Error 2
> make: *** [Makefile:141: all] Error 2
> ```
> 
> I have Open3D 0.9.0 installed in /usr/local/bin/cmake/, and Open3D/Geometry/KDTreeFlann.h actually exists in /usr/local/bin/cmake/include, so I guess the problem is with CMake not seeing correct Open3D path. Do you have any ideas on how to fix this?
> 
> **Update 2:** Solved traj_sampler issue by changing path to my Open3D installation in CMakeLists.txt of traj_sampler.
> 
> **However, rviz issue is still present**

Rename #include <Open3D/Geometry/KDTreeFlann.h> to "open/geometry//KDTreeFlann.h" maybe ok. Please check the Open3d installed path to find the include files such as "/usr/local/bin/cmake/open3d".
You need to rename all the include file of open3d according to your installed path.
Rviz issue is tricky.

den250400, 2021-11-05:
> Rename #include <Open3D/Geometry/KDTreeFlann.h> to "open/geometry//KDTreeFlann.h" maybe ok. Please check the Open3d installed path to find the include files such as "/usr/local/bin/cmake/open3d". [...]

How did you solve rviz issue? You already had version 1.13?

MaxTianyy, 2021-11-05:
> How did you solve rviz issue? You already had version 1.13?

NO~~~. We are in the same boat. I even try to use ubuntu 18.04 & ROS Melodic , but fail too.

HarrySoteriou, 2021-11-05:
I had some issues with octomap and solved it through here https://darienmt.com/autonomous-flight/2018/11/15/installing-ethz-rotors.html . maybe some issues can be fixed from here and others by downloading some packages you dont have from apt search ros-rviz

den250400, 2021-11-05:
> I had some issues with octomap and solved it through here https://darienmt.com/autonomous-flight/2018/11/15/installing-ethz-rotors.html . maybe some issues can be fixed from here and others by downloading some packages you dont have from apt search ros-rviz
> 
> for rviz check if you have the latest version of OGRE and qt5: sudo apt install qt5-default sudo apt-get install ogre-2.2
> 
> for traj_sampler **You correctly said** Rename #include <Open3D/Geometry/KDTreeFlann.h> to "open3d/geometry//KDTreeFlann.h" **but also** change the other 2 include to #include <open3d/geometry/PointCloud.h> #include <open3d/io/PointCloudIO.h>

I already solved traj_sampler issue by changing path to my Open3D installation in CMakeLists.txt of traj_sampler. No more actions were required. 
The only remaining problem is rviz. Would be much easier if ros maintainers stored older versions of packages in their apt repo...

By the way, did anyone try to build rviz from sources?

HarrySoteriou, 2021-11-08:
I tried building rviz from source just like open3d:
1) git clone https://github.com/ros-visualization/rviz
2) in this directory there is a link with some dependencies http://tango.freedesktop.org/releases/ (icon-naming-utils and tango-item-theme) using apt search **insert-package-name** find and install the latest versions
3) follow steps 3-5 just like here http://www.open3d.org/docs/release/compilation.html#config
4) if you face some errors of this type https://forum.openwrt.org/t/python3-pip3-aarch64-help/74685 around 86% it might be because g++ and gcc compilers are of version 7 instead of >8. Follow this guide to install a newer gcc and g++ version and use those for compiling instead https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa
5) if you successfully manage to build and make install rviz then you should see it in your /include folder (mine is not there yet, I am working on it)

In addition, going through the PointCloudCommon documentation it seems that getCallbackQueue() depends on callback_queue_interface.h which is not present in the project
http://docs.ros.org/en/jade/api/rviz/html/c++/classrviz_1_1PointCloudCommon.html

I dont know if any of these are the solution I just thought I share what I found so far

yfaws, 2021-11-11:
I also have a problem with rviz and take an error as below. 

> Errors     << planning_ros_utils:make /home/yfaktas/agile_autonomy_ws/catkin_aa/logs/planning_ros_utils/build.make.002.log
> /home/yfaktas/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp: In constructor ‘planning_rviz_plugins::MapDisplay::MapDisplay()’:
> /home/yfaktas/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp:39:52: error: ‘class rviz::PointCloudCommon’ has no member named ‘getCallbackQueue’
>    update_nh_.setCallbackQueue(point_cloud_common_->getCallbackQueue());

> make[2]: *** [CMakeFiles/planning_rviz_plugins.dir/src/planning_rviz_plugins/map_display.cpp.o] Error 1
> make[1]: *** [CMakeFiles/planning_rviz_plugins.dir/all] Error 2

Is there anyone solved this problem ?

antonilo, 2021-11-12:
If your issues persist, I would recommend commenting away that part. It is anyway only used for visualizing the point cloud while executing the code. It is not a key part of the project.

iamrajee, 2021-11-25:
@den250400 
I'm facing the same issue on noetic. Did you manage to solve it?
```
Failed     << example_control:make                              [ Exited with code 2 ]                                                                                
Failed    <<< example_control                                   [ 1.5 seconds ]                                                                                       
______________________________________________________________________________________________________________________________________________________________________
Errors     << planning_ros_utils:make /home/rajendra/agile_autonomy_ws/catkin_aa/logs/planning_ros_utils/build.make.009.log                                           
/home/rajendra/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp: In constructor ‘planning_rviz_plugins::MapDisplay::MapDisplay()’:
/home/rajendra/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp:39:52: error: ‘class rviz::PointCloudCommon’ has no member named ‘getCallbackQueue’
   39 |   update_nh_.setCallbackQueue(point_cloud_common_->getCallbackQueue());
      |                                                    ^~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/planning_rviz_plugins.dir/build.make:160: CMakeFiles/planning_rviz_plugins.dir/src/planning_rviz_plugins/map_display.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:381: CMakeFiles/planning_rviz_plugins.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
cd /home/rajendra/agile_autonomy_ws/catkin_aa/build/planning_ros_utils; catkin build --get-env planning_ros_utils | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

```

den250400, 2021-11-25:
> @den250400 I'm facing the same issue on noetic. Did you manage to solve it? **error: ‘class rviz::PointCloudCommon’ has no member named ‘getCallbackQueue’**
> 
> ```
> Failed     << example_control:make                              [ Exited with code 2 ]                                                                                
> Failed    <<< example_control                                   [ 1.5 seconds ]                                                                                       
> ______________________________________________________________________________________________________________________________________________________________________
> Errors     << planning_ros_utils:make /home/rajendra/agile_autonomy_ws/catkin_aa/logs/planning_ros_utils/build.make.009.log                                           
> /home/rajendra/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp: In constructor ‘planning_rviz_plugins::MapDisplay::MapDisplay()’:
> /home/rajendra/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp:39:52: error: ‘class rviz::PointCloudCommon’ has no member named ‘getCallbackQueue’
>    39 |   update_nh_.setCallbackQueue(point_cloud_common_->getCallbackQueue());
>       |                                                    ^~~~~~~~~~~~~~~~
> make[2]: *** [CMakeFiles/planning_rviz_plugins.dir/build.make:160: CMakeFiles/planning_rviz_plugins.dir/src/planning_rviz_plugins/map_display.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:381: CMakeFiles/planning_rviz_plugins.dir/all] Error 2
> make: *** [Makefile:160: all] Error 2
> cd /home/rajendra/agile_autonomy_ws/catkin_aa/build/planning_ros_utils; catkin build --get-env planning_ros_utils | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
> ```

Just like @antonilo said, I commented away the line with getCallbackQueue, and the build succeeded

den250400, 2021-11-28:
After commenting away line with getCallbackQueue and solving a few other minor problems, **I have managed to launch the simulation environment and test_trajectory.py**

In order to help those who struggle to build this project, I tried my best to distill all my actions into this installation manual. It is supposed to work even on freshly-installed Ubuntu 20.04. However, I know that there might be flaws in this guide, so feel free to report them.

```bash
sudo apt-get update
sudo apt-get install git cmake g++

# Install NVidia drivers, CUDA 11.2 and cuDNN
# This can be quite painful, but this tutorial helped me a lot https://yakhyo.medium.com/cuda-11-2-installation-on-ubuntu-20-04-e83f7561ccc1
# Be careful with version compatibility of CUDA and nvidia driver!

# Install Open3D v0.9.0
git clone --recursive https://github.com/intel-isl/Open3D
git checkout v0.9.0
git submodule update --init --recursive
./util/scripts/install-deps-ubuntu.sh
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr/local/bin/cmake ..
make -j$(nproc)
sudo make install

# Install catkin
sudo apt-get install ros-noetic-catkin python3-catkin-tools

# Install ROS Noetic
# Don't forget to configure your Ubuntu repositories to allow "restricted," "universe," and "multiverse."
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt install curl # if you haven't already installed curl
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
sudo apt update
sudo apt install ros-noetic-desktop-full
source /opt/ros/noetic/setup.bash
echo "source /opt/ros/noetic/setup.bash" >> ~/.bashrc
source ~/.bashrc
sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential
sudo apt install python3-rosdep
sudo rosdep init
rosdep update

# Install agile autonomy
# At that moment, you must create an ssh key and bind it to your github. Otherwise, clone and submodule update commands will fail. Check https://docs.github.com/en/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent for details

sudo apt-get install g++-7 gcc-7 # agile autonomy requires old compilers for some reason
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-7 100
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 100

export ROS_VERSION=noetic
mkdir agile_autonomy_ws
cd agile_autonomy_ws
export CATKIN_WS=./catkin_aa
mkdir -p $CATKIN_WS/src
cd $CATKIN_WS
catkin init
catkin config --extend /opt/ros/$ROS_VERSION
catkin config --merge-devel
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3
cd src
git clone git@github.com:uzh-rpg/agile_autonomy.git

sudo apt install python3-vcstool
vcs-import < agile_autonomy/dependencies.yaml
cd rpg_mpl_ros
git submodule update --init --recursive

# Install extra dependencies (might need more depending on your OS)
sudo apt-get install -y libqglviewer-dev-qt5

# Install external libraries for rpg_flightmare
sudo apt install -y libzmqpp-dev libeigen3-dev libglfw3-dev libglm-dev

# Install dependencies for rpg_flightmare renderer
sudo apt install -y libvulkan1 vulkan-utils gdb

# Install other dependencies not listed in original installation guide
sudo apt-get install libsdl-dev libsdl-image1.2-dev ros-noetic-octomap ros-noetic-octomap-msgs ros-noetic-octomap-ros

# Add environment variables (Careful! Modify path according to your local setup)
echo 'export RPGQ_PARAM_DIR=/home/denis/agile_autonomy/catkin_aa/src/rpg_flightmare' >> ~/.bashrc

# IMPORTANT: Before building the project, manually change several files in this project as described below in "File modifications" section

# Open another terminal window and build the project
sudo catkin build
# Setup the project
cd ..
. ../devel/setup.bash

# Create your learning environment
# You will also need to install Anaconda (You may install the latest version - it looks like you don't necessarily need Anaconda==v4.8.3 to run this project)
roscd planner_learning
conda create --name tf_24 python=3.7
conda activate tf_24
conda install tensorflow-gpu
pip install rospkg==1.2.3,pyquaternion,open3d,opencv-python
```

File modifications:

- catkin_aa/src/rpg_mpl_ros/open3d_conversions/CMakeLists.txt
replace line 13 with `find_package(Open3D HINTS /usr/local/bin/cmake/)`

- catkin_aa/src/rpg_mpl_ros/planning_ros_utils/src/planning_rviz_plugins/map_display.cpp
Comment out line 39

- catkin_aa/src/agile_autonomy/data_generation/traj_sampler/CMakeLists.txt
replace line 12 with `find_package(Open3D HINTS /usr/local/bin/cmake/)`

Hope it helps!

iamrajee, 2021-11-28:
@den250400, You are a lifesaver! Thank you for providing the detailed instructions.

yukaryote, 2021-11-29:
@den250400 thank you for the instructions! I was able to get 60/70 packages built when running `catkin build` but encountered this error when building traj_sampler: 
```
Errors     << traj_sampler:make /home/improbable/agile_autonomy_ws/catkin_aa/logs/traj_sampler/build.make.007.log                                                                                
In file included from /home/improbable/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/traj_sampler/src/kdtree.cpp:1:0:
/home/improbable/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/traj_sampler/include/traj_sampler/kdtree.h:5:10: fatal error: Open3D/Geometry/KDTreeFlann.h: No such file or directory
 #include <Open3D/Geometry/KDTreeFlann.h>
```
I tried changing `Open3D` to `open3d` in the include statement but that didn't work. Have you encountered something similar?

iamrajee, 2021-11-29:
@yukaryote I too had got this error, and I fixed by changing
`#include <Open3D/Geometry/KDTreeFlann.h>`
to 
`#include "open3d/geometry/KDTreeFlann.h"`

ZhangHaley, 2021-12-02:
All of the Open3D-related issues occurred due to different versions and installed positions; the authors use Open3D v0.9.0. and installed the package in the default folder such as "/usr/local/lib/cmake", the "include" path is "/usr/local/include/Open3D". When you are installing open3d, try:
`cmake ..` 
`make -j$(nproc)`
`sudo make install`
Then you don't need to modify the files of Open3D

linzgood, 2021-12-06:
linzgood@linzgood-ThinkPad-P15v-Gen-1:~/agile_autonomy_ws/catkin_aa$ sudo catkin build 
[sudo] password for linzgood: 
-------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/linzgood/agile_autonomy_ws/catkin_aa
-------------------------------------------------------------------------------
Build Space:        [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /home/linzgood/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------
[build] Found '73' packages in 0.0 seconds.                                                                                                                                                              
[build] Package table is up to date.                                                                                                                                                                     
Starting  >>> catkin_boost_python_buildtool                                                                                                                                                              
Starting  >>> catkin_simple                                                                                                                                                                              
Starting  >>> decomp_util                                                                                                                                                                                
Starting  >>> mav_msgs                                                                                                                                                                                   
Starting  >>> mav_state_machine_msgs                                                                                                                                                                     
Starting  >>> mav_system_msgs                                                                                                                                                                            
Starting  >>> motion_primitive_library                                                                                                                                                                   
Starting  >>> open3d_conversions                                                                                                                                                                         
Starting  >>> rotors_comm                                                                                                                                                                                
Starting  >>> rotors_description                                                                                                                                                                         
Starting  >>> rotors_evaluation                                                                                                                                                                          
Starting  >>> rqt_rotors                                                                                                                                                                                 
Finished  <<< catkin_boost_python_buildtool                     [ 0.1 seconds ]                                                                                                                          
Finished  <<< mav_state_machine_msgs                            [ 0.3 seconds ]                                                                                                                          
Finished  <<< rotors_evaluation                                 [ 0.1 seconds ]                                                                                                                          
Finished  <<< mav_system_msgs                                   [ 0.3 seconds ]                                                                                                                          
Finished  <<< catkin_simple                                     [ 0.1 seconds ]                                                                                                                          
[... cut ...]
[build]   Abandoned: None.                                                                                                                                                                               
[build]   Failed:    1 packages failed.                                                                                                                                                                  
[build] Runtime: 8.7 seconds total.                                                                                                                                                                      
linzgood@linzgood-ThinkPad-P15v-Gen-1:~/agile_autonomy_ws/catkin_aa$

linzgood, 2021-12-06:
> After commenting away line with getCallbackQueue and solving a few minor problems, **I have managed to launch the simulation environment and test_trajectory.py**
> 
> In order to help those who struggle to build this project, I tried my best to distill all my actions into this installation manual. It is supposed to work even on freshly-installed Ubuntu 20.04. However, I know that there might be flaws in this guide, so feel free to report them.
> 
> ```shell
> sudo apt-get update
> sudo apt-get install git cmake g++
> 
> # Install NVidia drivers, CUDA 11.2 and cuDNN
> # This can be quite painful, but this tutorial helped me a lot https://yakhyo.medium.com/cuda-11-2-installation-on-ubuntu-20-04-e83f7561ccc1
> # Be careful with version compatibility of CUDA and nvidia driver!
> 
> # Install Open3D v0.9.0
> git clone --recursive https://github.com/intel-isl/Open3D
> git checkout v0.9.0
> git submodule update --init --recursive
> ./util/scripts/install-deps-ubuntu.sh
> mkdir build
> cd build
> cmake -DCMAKE_INSTALL_PREFIX=/usr/local/bin/cmake ..
> make -j$(nproc)
> sudo make install
> 
> # Install catkin
> sudo apt-get install ros-noetic-catkin python3-catkin-tools
> 
> # Install ROS Noetic
> # Don't forget to configure your Ubuntu repositories to allow "restricted," "universe," and "multiverse."
> sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
> sudo apt install curl # if you haven't already installed curl
> curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
> sudo apt update
> sudo apt install ros-noetic-desktop-full
> source /opt/ros/noetic/setup.bash
> echo "source /opt/ros/noetic/setup.bash" >> ~/.bashrc
> source ~/.bashrc
> sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential
> sudo apt install python3-rosdep
> sudo rosdep init
> rosdep update
> 
> # Install agile autonomy
> # At that moment, you must create an ssh key and bind it to your github. Otherwise, clone and submodule update commands will fail. Check https://docs.github.com/en/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent for details
> 
> sudo apt-get install g++-7 gcc-7 # agile autonomy requires old compilers for some reason
> sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-7 100
> sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 100
> 
> export ROS_VERSION=noetic
> mkdir agile_autonomy_ws
> cd agile_autonomy_ws
> export CATKIN_WS=./catkin_aa
> mkdir -p $CATKIN_WS/src
> cd $CATKIN_WS
> catkin init
> catkin config --extend /opt/ros/$ROS_VERSION
> catkin config --merge-devel
> catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3
> cd src
> git clone git@github.com:uzh-rpg/agile_autonomy.git
> 
> sudo apt install python3-vcstool
> vcs-import < agile_autonomy/dependencies.yaml
> cd rpg_mpl_ros
> git submodule update --init --recursive
> 
> # Install extra dependencies (might need more depending on your OS)
> sudo apt-get install -y libqglviewer-dev-qt5
> 
> # Install external libraries for rpg_flightmare
[... 231 lines cut ...]

CMake Error at /home/linzgood/agile_autonomy_ws/catkin_aa/devel/share/open3d_conversions/cmake/open3d_conversionsConfig.cmake:173 (message):
  Project 'mpl_test_node' tried to find library 'Open3D'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'open3d_conversions'? Did you find_package() it before the subdirectory
  containing its code is included?
Call Stack (most recent call first):
  CMakeLists.txt:27 (find_package)

cd /home/linzgood/agile_autonomy_ws/catkin_aa/build/mpl_test_node; catkin build --get-env mpl_test_node | catkin env -si  /usr/bin/cmake /home/linzgood/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/linzgood/agile_autonomy_ws/catkin_aa/devel -DCMAKE_INSTALL_PREFIX=/home/linzgood/agile_autonomy_ws/catkin_aa/install -DCMAKE_BUILD_TYPE=Release; cd -

.........................................................................................................................................................................................................
Failed     << mpl_test_node:cmake                               [ Exited with code 1 ]                                                                                                                   
Failed    <<< mpl_test_node                                     [ 0.7 seconds ]                                                                                                                          
Finished  <<< traj_sampler                                      [ 0.4 seconds ]                                                                                                                          
[build] Summary: 72 of 73 packages succeeded.                                                                                                                                                            
[build]   Ignored:   None.                                                                                                                                                                               
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                  
[build] Runtime: 8.7 seconds total.                                                                                                                                                                      
linzgood@linzgood-ThinkPad-P15v-Gen-1:~/agile_autonomy_ws/catkin_aa$ 
``
![Screenshot from 2021-12-06 10-38-25](https://user-images.githubusercontent.com/10878749/144782118-c671964d-afcf-4125-8fd7-130636205c97.png)
`

linzgood, 2021-12-06:
```
linzgood@linzgood-ThinkPad-P15v-Gen-1:~/agile_autonomy_ws/catkin_aa$ sudo catkin build 
[sudo] password for linzgood: 
-------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/linzgood/agile_autonomy_ws/catkin_aa
-------------------------------------------------------------------------------
Build Space:        [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /home/linzgood/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /home/linzgood/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------
[build] Found '73' packages in 0.0 seconds.                                                                                                                                                              
[build] Package table is up to date.                                                                                                                                                                     
Starting  >>> catkin_boost_python_buildtool                                                                                                                                                              
Starting  >>> catkin_simple                                                                                                                                                                              
Starting  >>> decomp_util                                                                                                                                                                                
Starting  >>> mav_msgs                                                                                                                                                                                   
Starting  >>> mav_state_machine_msgs                                                                                                                                                                     
Starting  >>> mav_system_msgs                                                                                                                                                                            
Starting  >>> motion_primitive_library                                                                                                                                                                   
Starting  >>> open3d_conversions                                                                                                                                                                         
Starting  >>> rotors_comm                                                                                                                                                                                
Starting  >>> rotors_description                                                                                                                                                                         
Starting  >>> rotors_evaluation                                                                                                                                                                          
Starting  >>> rqt_rotors                                                                                                                                                                                 
Finished  <<< catkin_boost_python_buildtool                     [ 0.1 seconds ]                                                                                                                          
Finished  <<< mav_state_machine_msgs                            [ 0.3 seconds ]                                                                                                                          
Finished  <<< rotors_evaluation                                 [ 0.1 seconds ]                                                                                                                          
Finished  <<< mav_system_msgs                                   [ 0.3 seconds ]                                                                                                                          
Finished  <<< catkin_simple                                     [ 0.1 seconds ]                                                                                                                          
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                  
[build] Runtime: 8.7 seconds total.                                                                                                                                                                      
linzgood@linzgood-ThinkPad-P15v-Gen-1:~/agile_autonomy_ws/catkin_aa$ 
``
![Screenshot from 2021-12-06 10-38-25](https://user-images.githubusercontent.com/10878749/144782855-06423aaf-03f5-4db7-bad1-a426439e2b32.png)

Howardcl, 2021-12-09:
@linzgood have you sloved this problem?I have got the same problem

linzgood, 2021-12-09:
> @linzgood have you sloved this problem?I have got the same problem

I have not sloved this problem.

Howardcl, 2021-12-13:
> 

hello~ What version of cmake do you use ？

Howardcl, 2021-12-13:
@den250400   What version of cmake do you use ？

lilrookie76, 2021-12-22:
> 

thank you guy, but one last question, the best way to solve the rviz issure is commenting away that line, a little bit sad...

den250400, 2021-12-24:
> > @den250400
> 
> thank you guy, but one last question, the best way to solve the rviz issue is commenting away that line, a little bit sad...

Yep, the only solution we found was to comment away line with rviz. What's really sad is that ROS repositories store only the latest version of their packages, so you can't just `sudo apt-get install rviz=1.13.3`. You can try to build rviz from sources by cloning its repo and cheking out to v1.13.3, but rviz maintainers haven't provided any instructions on how to do it.

SoheilSalehian, 2021-12-24:
@linzgood @Howardcl:  I am running into this following Den's approach. Were you able to figure out how to go past that error?

lilrookie76, 2021-12-25:
> > > @den250400
> > 
> > 
> > thank you guy, but one last question, the best way to solve the rviz issue is commenting away that line, a little bit sad...
> 
> Yep, the only solution we found was to comment away the line with rviz. What's really sad is that ROS repositories store only the latest version of their packages, so you can't just `sudo apt-get install ros-noetic-rviz=1.13.3`. You can try to build rviz from sources by cloning its repo and cheking out to v1.13.3, but rviz maintainers haven't provided any instructions on how to do it.

Actually, I tried to fix this problem by editing that line to adapt to the latest version of rviz and its cpp api, but I found that the document of the ros related package like

> > > @den250400
> > 
> > 
> > thank you guy, but one last question, the best way to solve the rviz issue is commenting away that line, a little bit sad...
> 
> Yep, the only solution we found was to comment away the line with rviz. What's really sad is that ROS repositories store only the latest version of their packages, so you can't just `sudo apt-get install ros-noetic-rviz=1.13.3`. You can try to build rviz from sources by cloning its repo and cheking out to v1.13.3, but rviz maintainers haven't provided any instructions on how to do it.

Actually, I tried to fix this problem by editing that line of code to adapt to the latest version of rviz and it's cpp API, but I found that the official document of ros related package API is too brief... and I must search and read more tutorials to edit the code, so it would be a good idea to comment away it at this time..

lilrookie76, 2021-12-25:
@SoheilSalehian @linzgood @Howardcl 
we are on the same boat...

linzgood, 2021-12-26:
> @linzgood have you sloved this problem?I have got the same problem

  /devel/share/open3d_conversions/cmake/open3d_conversionsConfig.cmake，line 157，added Open3D path：foreach(path  *;/home/yourname/open3d_install/lib)

linzgood, 2021-12-26:
> @linzgood @Howardcl: I am running into this following Den's approach. Were you able to figure out how to go past that error?
> 
> `CMake Error at /home/linzgood/agile_autonomy_ws/catkin_aa/devel/share/open3d_conversions/cmake/open3d_conversionsConfig.cmake:173 (message): Project 'mpl_test_node' tried to find library 'Open3D'. The library is neither a target nor built/installed properly. Did you compile project 'open3d_conversions'? Did you find_package() it before the subdirectory containing its code is included? Call Stack (most recent call first): CMakeLists.txt:27 (find_package)`

/devel/share/open3d_conversions/cmake/open3d_conversionsConfig.cmake，line 157，added Open3D path：foreach(path *;/home/yourname/open3d_install/lib)

erenleicter, 2022-01-10:
> /devel/share/open3d_conversions/cmake/open3d_conversionsConfig.cmake，line 157，added Open3D path：foreach(path *;/home/yourname/open3d_install/lib)

since I followed Den's instructions, I added the following Open3D path: foreach(**;/usr/local/bin/cmake/lib)

fjodorlundgren, 2022-03-27:
> I think that for your version of open3d you need to rename the include file of open3d to `#include "open3d/Open3D.h" `. An example is here: https://github.com/isl-org/open3d-cmake-find-package/blob/master/Draw.cpp#L29.
> 
> Hope it helps!

Can somebody please explain me (as a newbie) what the exact steps to follow are? I've got the same problems

mw9385, 2022-05-26:
@den250400  Hi, I followed your commands and succeeded in building. However, after launching `python dagger_training.py --settings_file=config/dagger_settings.yaml`, the drone does not move and the simulation is finished. Do you have any ideas for this? 

Thanks you.

kielerrr, 2022-10-23:
@den250400 fixed a few items on your awesome install script 

cmake -DCMAKE_INSTALL_PREFIX=/usr/local/~~bin~~/cmake ..
cmake -DCMAKE_INSTALL_PREFIX=/usr/local/lib/cmake ..

replace line 13 with find_package(Open3D HINTS /usr/local/~~bin~~/cmake/)
line 13 with find_package(Open3D HINTS /usr/local/lib/cmake/)

 replace line 12 with find_package(Open3D HINTS /usr/local/~~bin~~/cmake/)
 line 12 with find_package(Open3D HINTS /usr/local/lib/cmake/)

Glanfaloth, 2022-10-25:
In case someone run into
```CMake Error at 3rdparty/cmake_install.cmake:54 (file):
  file INSTALL cannot make directory
  "/usr/local/bin/cmake/include/Open3D/3rdparty/Eigen": Not a directory.
Call Stack (most recent call first):
  cmake_install.cmake:47 (include)
````
while running `sudo make install`, you can try `make install-pip-package`

kielerrr, 2022-10-25:
> 

I had the same issue but found it is only a typo on the install script. The directory `/usr/local/bin/cmake/include/Open3D` should be `/usr/local/lib/cmake/include/Open3D`

Glanfaloth, 2022-10-25:
> 

@kielerrr Where is the install script? Thank you!

kielerrr, 2022-10-25:
Its in this thread
 [here](https://github.com/uzh-rpg/agile_autonomy/issues/10#issuecomment-981095386)

NIHALNAVALE, 2023-03-14:
`Errors     << open3d_conversions:make /home/nihal/agile_autonomy_ws/catkin_aa/logs/open3d_conversions/build.make.007.log                                                                                                                                                         
In file included from /home/nihal/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/src/open3d_conversions.cpp:15:0:
/home/nihal/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19:10: fatal error: open3d/Open3D.h: No such file or directory
 #include <open3d/Open3D.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/open3d_conversions.dir/build.make:63: CMakeFiles/open3d_conversions.dir/src/open3d_conversions.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:305: CMakeFiles/open3d_conversions.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/nihal/agile_autonomy_ws/catkin_aa/build/open3d_conversions; catkin build --get-env open3d_conversions | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

.................................................................................................................................................................................................................................................................................
Failed     << open3d_conversions:make                           [ Exited with code 2 ]                                                                                                                                                                                           
Failed    <<< open3d_conversions                                [ 0.1 seconds ]                                                                                                                                                                                                  
Abandoned <<< assimp_catkin                                     [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< catkin_boost_python_buildtool_test                [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< decomp_ros_msgs                                   [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< decomp_ros_utils                                  [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< decomp_test_node                                  [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< eigen_catkin                                      [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< flightrender                                      [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< gflags_catkin                                     [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< glog_catkin                                       [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< eigen_checks                                      [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< mav_comm                                          [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< minimum_jerk_trajectories                         [ Unrelated job failed ]                                                                                                                                                                                         
Abandoned <<< minkindr                                          [ Unrelated job failed ]                                                                                                                                                                                         
[... cut ...]
Finished  <<< rotors_control                                    [ 0.2 seconds ]                                                                                                                                                                                                  
Finished  <<< motion_primitive_library                          [ 0.4 seconds ]                                                                                                                                                                                                  
Finished  <<< decomp_util                                       [ 0.2 seconds ]    `

How do you solve the open3d issue?

zhw970623, 2024-12-02:
I was trying to build this project recently, but a problem that had never occurred before occurred, which made me desperate.

`________________________________________________________________________________________________________________________________________________________
Errors     << rotors_gazebo_plugins:make /home/zhw/agile_autonomy_ws/catkin_aa/logs/rotors_gazebo_plugins/build.make.002.log                            
In file included from /usr/include/ignition/math6/gz/math/Spline.hh:22:0,
                 from /usr/include/ignition/math6/ignition/math/Spline.hh:18,
                 from /usr/include/gazebo-11/gazebo/common/Animation.hh:22,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:5,
                 from /home/zhw/agile_autonomy_ws/catkin_aa/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_bag_plugin.h:28,
                 from /home/zhw/agile_autonomy_ws/catkin_aa/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:22:
/usr/include/ignition/math6/gz/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/include/ignition/math6/gz/math/Helpers.hh:1062:28: error: ‘chrono_literals’ is not a namespace-name
       using namespace std::chrono_literals;
                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/gz/math/Helpers.hh:1062:43: error: expected namespace-name before ‘;’ token
       using namespace std::chrono_literals;
                                           ^
/usr/include/ignition/math6/gz/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/include/ignition/math6/gz/math/Helpers.hh:1104:28: error: ‘chrono_literals’ is not a namespace-name
       using namespace std::chrono_literals;
                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/gz/math/Helpers.hh:1104:43: error: expected namespace-name before ‘;’ token
       using namespace std::chrono_literals;
                                           ^
/usr/include/ignition/math6/gz/math/Helpers.hh:1105:56: error: unable to find numeric literal operator ‘operator""s’
       std::chrono::steady_clock::time_point timePoint{-1s};
                                                        ^~
/usr/include/ignition/math6/gz/math/Helpers.hh:1105:56: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/ignition/math6/gz/math/Helpers.hh:1105:58: error: no matching function for call to ‘std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(<brace-enclosed initializer list>)’
       std::chrono::steady_clock::time_point timePoint{-1s};
                                                          ^
In file included from /usr/include/ignition/math6/gz/math/Helpers.hh:21:0,
                 from /usr/include/ignition/math6/gz/math/Spline.hh:22,
                 from /usr/include/ignition/math6/ignition/math/Spline.hh:18,
                 from /usr/include/gazebo-11/gazebo/common/Animation.hh:22,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:5,
                 from /home/zhw/agile_autonomy_ws/catkin_aa/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_bag_plugin.h:28,
                 from /home/zhw/agile_autonomy_ws/catkin_aa/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp:22:
/usr/include/c++/7/chrono:633:14: note: candidate: template<class _Dur2, class> constexpr std::chrono::time_point<_Clock, _Dur>::time_point(const std::chrono::time_point<_Clock, _Dur2>&)
    constexpr time_point(const time_point<clock, _Dur2>& __t)
              ^~~~~~~~~~
/usr/include/c++/7/chrono:633:14: note:   template argument deduction/substitution failed:
/usr/include/c++/7/chrono:626:21: note: candidate: constexpr std::chrono::time_point<_Clock, _Dur>::time_point(const duration&) [with _Clock = std::chrono::_V2::steady_clock; _Dur = std::chrono::duration<long int, std::ratio<1, 1000000000> >; std::chrono::time_point<_Clock, _Dur>::duration = std::chrono::duration<long int, std::ratio<1, 1000000000> >]
  constexpr explicit time_point(const duration& __dur)
                     ^~~~~~~~~~
/usr/include/c++/7/chrono:626:21: note:   conversion of argument 1 would be ill-formed:
/usr/include/c++/7/chrono:623:12: note: candidate: constexpr std::chrono::time_point<_Clock, _Dur>::time_point() [with _Clock = std::chrono::_V2::steady_clock; _Dur = std::chrono::duration<long int, std::ratio<1, 1000000000> >]
  constexpr time_point() : __d(duration::zero())
            ^~~~~~~~~~
/usr/include/c++/7/chrono:623:12: note:   candidate expects 0 arguments, 1 provided
/usr/include/c++/7/chrono:616:14: note: candidate: constexpr std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(const std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >&)
       struct time_point
              ^~~~~~~~~~
/usr/include/c++/7/chrono:616:14: note:   conversion of argument 1 would be ill-formed:
/usr/include/c++/7/chrono:616:14: note: candidate: constexpr std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(std::chrono::time_point<std::chrono::_V2 [... cut ...]
[build]   Warnings:  None.                                                                                                                              
[build]   Abandoned: 32 packages were abandoned.                                                                                                        
[build]   Failed:    1 packages failed.                                                                                                                 
[build] Runtime: 17.4 seconds total.                                          `

hcws, 2025-08-24:
> sudo make install

it show errors ： 
```
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/bin/cmake/include/Open3D/3rdparty/Eigen
CMake Error at 3rdparty/cmake_install.cmake:54 (file):
  file INSTALL cannot make directory
  "/usr/local/bin/cmake/include/Open3D/3rdparty/Eigen": Not a directory.
Call Stack (most recent call first):
  cmake_install.cmake:47 (include)

make: *** [Makefile:100: install] Error 1
```