uzh-rpg/agile_autonomy #12: link error with generate_labels in traj_sampler

WhiteCri, 2021-11-10:
Hi, I'm trying to build this project in docker with ubuntu 20.04, gcc/g++7.5 installed.
I struggle to build this project a lot, but still I failed to build generate_labels node in traj_sampler pkg.
Below is the error I got:

```                                                                
root@tw:~/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/traj_sampler# catkin build traj_sampler 
---------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /root/agile_autonomy_ws/catkin_aa
---------------------------------------------------------------------------------------------
Build Space:        [exists] /root/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /root/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /root/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /root/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /root/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
---------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------------------------------
[build] Found '70' packages in 0.0 seconds.                                                             
[build] Package table is up to date.                                                                    
Starting  >>> catkin_simple                                                                             
Finished  <<< catkin_simple                               [ 0.1 seconds ]                               
Starting  >>> eigen_catkin                                                                              
Starting  >>> gflags_catkin                                                                             
Starting  >>> minimum_jerk_trajectories                                                                 
Starting  >>> quadrotor_msgs                                                                            
Finished  <<< minimum_jerk_trajectories                   [ 0.1 seconds ]                               
Finished  <<< gflags_catkin                               [ 0.1 seconds ]                               
Starting  >>> glog_catkin                                                                               
Finished  <<< eigen_catkin                                [ 0.1 seconds ]                               
Finished  <<< quadrotor_msgs                              [ 0.1 seconds ]                               
Starting  >>> quadrotor_common                                                                          
Finished  <<< glog_catkin                                 [ 0.1 seconds ]                               
Starting  >>> eigen_checks                                                                              
Finished  <<< quadrotor_common                            [ 0.1 seconds ]                               
Starting  >>> polynomial_trajectories                                                                   
Starting  >>> position_controller                                                                       
Starting  >>> state_predictor                                                                           
Finished  <<< eigen_checks                                [ 0.1 seconds ]                               
Starting  >>> minkindr                                                                                  
Finished  <<< polynomial_trajectories                     [ 0.1 seconds ]                               
Finished  <<< state_predictor                             [ 0.1 seconds ]                               
Finished  <<< position_controller                         [ 0.1 seconds ]                               
Finished  <<< minkindr                                    [ 0.1 seconds ]                               
Starting  >>> rpg_common                                                                                
Finished  <<< rpg_common                                  [ 0.1 seconds ]                               
Starting  >>> trajectory_generation_helper                                                              
Finished  <<< trajectory_generation_helper                [ 0.1 seconds ]                               
Starting  >>> autopilot                                                                                 
Finished  <<< autopilot                                   [ 0.1 seconds ]                               
Starting  >>> agile_autonomy_utils                                                                      
Starting  >>> rpg_mpc                                                                                   
Finished  <<< rpg_mpc                                     [ 0.2 seconds ]                               
Finished  <<< agile_autonomy_utils                        [ 0.1 seconds ]                               
Starting  >>> traj_sampler                                                                              
________________________________________________________________________________________________________
[... 120 lines cut ...]
make[2]: *** [CMakeFiles/generate_label.dir/build.make:410: /root/agile_autonomy_ws/catkin_aa/devel/lib/traj_sampler/generate_label] Error 1
make[1]: *** [CMakeFiles/Makefile2:1730: CMakeFiles/generate_label.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /root/agile_autonomy_ws/catkin_aa/build/traj_sampler; catkin build --get-env traj_sampler | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

........................................................................................................
Failed     << traj_sampler:make                           [ Exited with code 2 ]                        
Failed    <<< traj_sampler                                [ 1.4 seconds ]                               
[build] Summary: 17 of 18 packages succeeded.                                                           
[build]   Ignored:   52 packages were skipped or are blacklisted.                                       
[build]   Warnings:  1 packages succeeded with warnings.                                                
[build]   Abandoned: None.                                                                              
[build]   Failed:    1 packages failed.                                                                 
[build] Runtime: 2.8 seconds total.                                                                     
root@tw:~/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/traj_sampler#
```

I checked or tried below
- I checked gcc/g++ version, which were 7.5 each.
- I checked ${catkin_LIBRARIES} variable in CMakeLists.txt, and confirmed if necessary library(libroscpp.so) is included.
- I tried not using catkin_simple and tried ordinary CMakelists.txt format

Did anyone give me hinks about this?

Thanks in advance.

calmelo, 2021-11-18:
> Hi, I'm trying to build this project in docker with ubuntu 20.04, gcc/g++7.5 installed. I struggle to build this project a lot, but still I failed to build generate_labels node in traj_sampler pkg. [...]

I meet the problem too. Have you solved the problem?

WhiteCri, 2021-11-18:
not yet. Hope someone carry the solution...

racheraven, 2021-11-18:
I think it is a problem with the latest Open3D.
For detail, please refer to [https://github.com/isl-org/Open3D/issues/2286]().
Basically, I guess it's some part of Open3D crashes with ROS dependency.
just try 
```
git clone --recursive https://github.com/intel-isl/Open3D
cd Open3D && source util/scripts/install-deps-ubuntu.sh
mkdir build && cd build
cmake -DBUILD_EIGEN3=ON -DBUILD_GLEW=ON -DBUILD_GLFW=ON -DBUILD_JSONCPP=ON -DBUILD_PNG=ON -DGLIBCXX_USE_CXX11_ABI=ON -DPYTHON_EXECUTABLE=/usr/bin/python -DBUILD_UNIT_TESTS=ON ..
make -j4
sudo make install
```
when installing Open3D, as [https://github.com/ntnu-arl/open3d_ros#open3d]() suggests.

I have resolved the linking problem in this issue, though I haven't test the whole project yet.

I guess the authors have used an older version of Open3D, which doesn't have this linking issue. May @antonilo can inform us more about library version they used?

calmelo, 2021-11-18:
> make -j4

The authors use Open3D 0.9.0.0, I will try your method. Thanks!

calmelo, 2021-11-18:
> @WhiteCri I think it is a problem with the latest Open3D. For detail, please refer to https://github.com/isl-org/Open3D/issues/2286. Basically, I guess it's some part of Open3D crashes with ROS dependency. just try
> 
> ```
> git clone --recursive https://github.com/intel-isl/Open3D
> cd Open3D && source util/scripts/install-deps-ubuntu.sh
> mkdir build && cd build
> cmake -DBUILD_EIGEN3=ON -DBUILD_GLEW=ON -DBUILD_GLFW=ON -DBUILD_JSONCPP=ON -DBUILD_PNG=ON -DGLIBCXX_USE_CXX11_ABI=ON -DPYTHON_EXECUTABLE=/usr/bin/python -DBUILD_UNIT_TESTS=ON ..
> make -j4
> sudo make install
> ```
> 
> when installing Open3D, as https://github.com/ntnu-arl/open3d_ros#open3d suggests.
> 
> I am using Ubuntu 18.04, g++/gcc 7.5.0, and I have resolved the linking problem in this issue, though I haven't tested the whole project yet.
> 
> I guess the authors used an older version of Open3D, which doesn't have this linking issue. Maybe @antonilo can inform us more about library version they used?

Thank you for your reply. The solution is useful. I have resolved the problem.

WhiteCri, 2021-11-23:
@racheraven's advice with Open3D worked fine. Super nice. Thanks!

closed 2021-11-23 by WhiteCri

fishhide, 2025-09-10:
原因是traj_sampler包需要链接glob，可以在traj_sampler包的CMakeLists.txt下找到```target_link_libraries(kdtree ${Open3D_LIBRARIES}
        OpenMP::OpenMP_CXX)```，修改为```target_link_libraries(kdtree ${Open3D_LIBRARIES}
        OpenMP::OpenMP_CXX glog)```，然后catkin build就行了。其它包如mpl_test_node好像也有这个问题，同样找到其下CMakeLists.txt修改为```target_link_libraries(ellipsoid_planner_node ${MOTION_PRIMITIVE_LIBRARY_LIBRARIES} ${Open3D_LIBRARIES} OpenMP::OpenMP_CXX
        stdc++fs glog)```。

fishhide, 2025-09-10:
原因是traj_sampler包需要链接glog库，修改src/agile_autonomy/data_generation/traj_sampler/CMakeLists.txt的44行左右，
```
target_link_libraries(kdtree ${Open3D_LIBRARIES}
        OpenMP::OpenMP_CXX)
```
变成
```
target_link_libraries(kdtree ${Open3D_LIBRARIES}
        OpenMP::OpenMP_CXX glog)
```
类似mpl_test_node包也有这个问题，修改src/rpg_mpl_ros/mpl_test_node/CMakeLists.txt的68行左右
```
target_link_libraries(ellipsoid_planner_node ${MOTION_PRIMITIVE_LIBRARY_LIBRARIES} ${Open3D_LIBRARIES} OpenMP::OpenMP_CXX
        stdc++fs)
```
变成
```
target_link_libraries(ellipsoid_planner_node ${MOTION_PRIMITIVE_LIBRARY_LIBRARIES} ${Open3D_LIBRARIES} OpenMP::OpenMP_CXX
        stdc++fs glog)
```