uzh-rpg/agile_autonomy #18: [ INFO] [1638364145.246536777, 2.618000000]: Unity is ready! [ERROR] [1638364145.596009985, 2.793000000]: no kernel image is available for execution on the device 209

calmelo, 2021-12-01:
[ INFO] [1638364145.246536777, 2.618000000]: Unity is ready!
[ERROR] [1638364145.596009985, 2.793000000]: no kernel image is available for execution on the device 209

Has anyone meet the problem? My environment is RTX3080ti, CUDA11.3. The error happened when I use unity. I don't know the reason. How can I resolve the problem?

ZhangHaley, 2021-12-02:
Have you found a solution to the problem? I was in the same boat as you. @calmelo

calmelo, 2021-12-02:
> Have you found a solution to the problem? I was in the same boat as you. @calmelo

Sorry, I haven't. Have you resolved the problem?

ZhangHaley, 2021-12-04:
> > Have you found a solution to the problem? I was in the same boat as you. @calmelo
> 
> Sorry, I haven't. Have you resolved the problem?

My issue has been resolved. My operating system is Ubuntu 20.04, and my graphics card is an Nvidia Geforce GTX 3070. Initially, my graphics driver is nvidia-450 with CUDA11.0. However, gcc and g++ 7.5.0 are used in this project. They are incompatible with CUDA11.0. As a result, I updated my graphics driver to nvidia-470.86 and changed CUDA11.0 to CUDA10.1, which is compatible with gcc7.5. Then it works fine, except for the slow frame rate.

calmelo, 2021-12-06:
> Have you found a solution to the problem? I was in the same boat as you. @calmelo

I saw that you have resolved the problem by using CUDA10.1 from email. Why did you delete the comments? Is the method a error?

racheraven, 2021-12-06:
I believe it is the problem with the dir where you are running your .py that cause the problem.
Changing your dir to `agile_autonomy/planner_learning/ `before running the `test_trajectories.py` might help.
Don't forget to create a `agile_autonomy/data_generation/data`, where things like global traj, pointclound etc. is stored during testing.

racheraven, 2021-12-06:
This might be caused by your version of CUDA, tensorflow. Make sure that these versions are in accordance with each other. You may check this on NVIDIA's website.

Also, remember to change line 14 in your `agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt ` according to your GPU architecture. Similarly, refer to NVIDIA's website.

calmelo, 2021-12-06:
> This might be caused by your version of CUDA, tensorflow. Make sure that these versions are in accordance with each other. You may check this on NVIDIA's website.
>
> Also, remember to change line 14 [...]

Thanks for your reply, I will try to change line 14 in my agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt according to my GPU architecture. I didn't use the test_trajectories.py. I just run roslaunch agile_autonomy simulation.launch, and I meet the problem. So I think the error is not related to the tensorflow.

calmelo, 2021-12-06:
> This might be caused by your version of CUDA, tensorflow. Make sure that these versions are in accordance with each other. You may check this on NVIDIA's website.
>
> Also, remember to change line 14 [...]

Thanks for your solution. I resolved the problem by change line 14 in my agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt. Very thank you!

closed 2021-12-06 by calmelo

ZhangHaley, 2021-12-06:
> 

Could you tell me how did you change the CmakeLists.txt, thanks a lot.c

zerozerobot, 2022-04-19:
How did you change the CMakeLists.text. Thanks!

slyangxu, 2022-11-19:
> Thanks for your solution. I resolved the problem by change line 14 in my agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt. Very thank you!

Could you tell me how did you change the CmakeLists.txt, thanks a lot.c

Liviy-yu, 2023-03-14:
你好，请问是怎么修改呀？