uzh-rpg/agile_autonomy #23: Why python test_trajectories.py --settings_file=config/test_settings.yaml is so slow?

calmelo, 2021-12-06:
When I run python test_trajectories.py --settings_file=config/test_settings.yaml, it is so slow to upload the model. I need to wait for several minutes. Can someone tell me the reason?

ZhangHaley, 2021-12-06:
I have tried cuda11.4 it is more faster then other version.

ZhangHaley, 2021-12-06:
I have tried CUDA11.4. It is faster than other versions.

calmelo, 2021-12-06:
> I have tried CUDA11.4. It is faster than other versions.

The command of conda install tensorflow-gpu install the cuda10.1 default in tf_24 environment. How did you change the CUDA version？

NPU-yuhang, 2021-12-10:
You can install tensorflow2.4 by pip instead of conda

calmelo, 2021-12-13:
> You can install tensorflow2.4 by pip instead of conda

Did you mean that installing tensorflow2.4 by pip can install tensorflow2.4 with cuda11?

NPU-yuhang, 2021-12-13:
Yes, but you should install cuda11 from Nvidia developer website before installing the tensorflow2.4.