uzh-rpg/agile_autonomy #24: Environment not rendering after roslaunch agile_autonomy simulation.launch

yukaryote, 2021-12-06:
When I run `roslaunch agile_autonomy simulation.launch`, I get the following error:
```
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
[hummingbird/agile_autonomy-9] process has died [pid 296, exit code -6, cmd /home/improbable/agile_autonomy_ws/catkin_aa/devel/lib/agile_autonomy/agile_autonomy completed_global_plan:=/test_primitive/completed_planning __name:=agile_autonomy __log:=/home/improbable/.ros/log/3e1f4066-56e9-11ec-a21d-0c9d92c3307d/hummingbird-agile_autonomy-9.log].
log file: /home/improbable/.ros/log/3e1f4066-56e9-11ec-a21d-0c9d92c3307d/hummingbird-agile_autonomy-9*.log
[flight_render-20] process has died [pid 490, exit code -11, cmd /home/improbable/agile_autonomy_ws/catkin_aa/src/agile_autonomy/flightmare_agile_autonomy/flightrender/20201127/flightmare.x86_64 __name:=flight_render __log:=/home/improbable/.ros/log/3e1f4066-56e9-11ec-a21d-0c9d92c3307d/flight_render-20.log].
log file: /home/improbable/.ros/log/3e1f4066-56e9-11ec-a21d-0c9d92c3307d/flight_render-20*.log
```
This is causing the environment to not show up in Rviz (and probably something more important isn't working).

yuwei-wu, 2021-12-10:
Did you correctly do echo 'export RPGQ_PARAM_DIR=/home/<path/to/>catkin_aa/src/rpg_flightmare' >> ~/.bashrc and revise the path to your own path ?

I encountered the same issue and solve it by this. Hope it can help you

yukaryote, 2021-12-11:
That seemed to fix the std logic error thanks! However, I am now getting a "Spawn service failed. Exiting" error

yukaryote, 2021-12-12:
That resolved the std logic error, thanks! However, I get this spawn error now: 
```
[ INFO] [1639270888.563694457, 185673.449000000]: [/hummingbird/generate_label_7] Q_perception = 0
[INFO] [1639270888.564393, 185673.450000]: Spawn status: SpawnModel: Failure - entity already exists.
[ INFO] [1639270888.564634212, 185673.450000000]: [/hummingbird/generate_label_1] max_thrust = 15
[ INFO] [1639270888.565668468, 185673.450000000]: [/hummingbird/generate_label_5] R_thrust = 1
[ INFO] [1639270888.568337943, 185673.452000000]: [/hummingbird/agile_autonomy] R_pitchroll = 1
[ INFO] [1639270888.569367256, 185673.452000000]: [/hummingbird/generate_label_6] R_thrust = 1
[ INFO] [1639270888.570364779, 185673.453000000]: [/hummingbird/generate_label_2] max_bodyrate_xy = 5
[ INFO] [1639270888.571569237, 185673.454000000]: [/hummingbird/autopilot] emergency_land_duration = 4
[ INFO] [1639270888.572609102, 185673.454000000]: [/hummingbird/generate_label_3] input_cost_exponential = 0
[ INFO] [1639270888.573576423, 185673.454000000]: [/hummingbird/generate_label_4] state_cost_exponential = 0
[ INFO] [1639270888.574805373, 185673.455000000]: [/hummingbird/generate_label_7] R_thrust = 1
[ERROR] [1639270888.577265, 185673.456000]: Spawn service failed. Exiting.
[ INFO] [1639270888.578192178, 185673.457000000]: [/hummingbird/generate_label_5] R_pitchroll = 1
[ INFO] [1639270888.579423778, 185673.457000000]: [/hummingbird/generate_label_0] traj_len = 10
[ INFO] [1639270888.580420477, 185673.458000000]: [/hummingbird/agile_autonomy] R_yaw = 1
```

djmadden, 2024-03-24:
@yukaryote did you manage to resolve this issue? I am having the same problem. 

Running on Ubuntu 20.04 server build (arm64). I am also having issues where the program doesn't believe PyQt5 and PySide2 are installed, but it still loads `rviz`, except there is no environment rendering. 

If no response, I will raise as an issue.