uzh-rpg/agile_autonomy #26: open3d_conversions error when catkin build

Howardcl, 2021-12-13:
[build] Found '71' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> catkin_boost_python_buildtool                                    
Starting  >>> catkin_simple                                                    
Starting  >>> decomp_util                                                      
Starting  >>> mav_msgs                                                         
Starting  >>> mav_state_machine_msgs                                           
Starting  >>> mav_system_msgs                                                  
Starting  >>> motion_primitive_library                                         
Starting  >>> open3d_conversions                                               
Starting  >>> rotors_comm                                                      
Starting  >>> rotors_description                                               
Starting  >>> rotors_evaluation                                                
Starting  >>> rqt_rotors                                                       
Finished  <<< mav_state_machine_msgs                            [ 0.1 seconds ]
Finished  <<< mav_msgs                                          [ 0.1 seconds ]
Starting  >>> mav_planning_msgs                                                
Starting  >>> rotors_control                                                   
Starting  >>> rotors_hil_interface                                             
Starting  >>> rotors_joy_interface                                             
Finished  <<< rotors_evaluation                                 [ 0.1 seconds ]
Finished  <<< catkin_boost_python_buildtool                     [ 0.0 seconds ]
Finished  <<< mav_system_msgs                                   [ 0.1 seconds ]
Finished  <<< rqt_rotors                                        [ 0.1 seconds ]
Finished  <<< catkin_simple                                     [ 0.1 seconds ]
Starting  >>> assimp_catkin                                                    
Starting  >>> catkin_boost_python_buildtool_test                               
Starting  >>> decomp_ros_msgs                                                  
Starting  >>> eigen_catkin                                                     
Starting  >>> flightrender                                                     
Starting  >>> gflags_catkin                                                    
Starting  >>> minimum_jerk_trajectories                                        
Starting  >>> planning_ros_msgs                                                
Starting  >>> quad_launch_files                                                
Starting  >>> quadrotor_msgs                                                   
Starting  >>> rpg_single_board_io                                              
Starting  >>> rpgq_msgs                                                        
Finished  <<< rotors_comm                                       [ 0.1 seconds ]
Finished  <<< rotors_description                                [ 0.1 seconds ]
Finished  <<< rotors_control                                    [ 0.1 seconds ]
Finished  <<< rotors_hil_interface                              [ 0.1 seconds ]
Starting  >>> rotors_gazebo_plugins                                            
Finished  <<< mav_planning_msgs                                 [ 0.1 seconds ]
Starting  >>> mav_comm                                                         
Finished  <<< rotors_joy_interface                              [ 0.1 seconds ]
Finished  <<< decomp_ros_msgs                                   [ 0.1 seconds ]
Finished  <<< quad_launch_files                                 [ 0.1 seconds ]
Finished  <<< gflags_catkin                                     [ 0.1 seconds ]
Starting  >>> glog_catkin                                                      
_______________________________________________________________________________
Warnings   << open3d_conversions:check /home/pc205/agile_autonomy_ws/catkin_aa/logs/open3d_conversions/build.check.022.log
Searching open3D in /usr/local/bin/cmake/
Found OpenMP TRUE    0.9.0.0 OpenMP::OpenMP_CXX
Found Open3D 0.9.0.0
cd /home/pc205/agile_autonomy_ws/catkin_aa/build/open3d_conversions; catkin build --get-env open3d_conversions | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

...............................................................................
Finished  <<< open3d_conversions                                [ 0.5 seconds ]
Finished  <<< eigen_catkin                                      [ 0.1 seconds ]
Starting  >>> numpy_eigen                                                      
Finished  <<< catkin_boost_python_buildtool_test                [ 0.1 seconds ]
Finished  <<< rpgq_msgs                                         [ 0.1 seconds ]
Starting  >>> rpgq_common                                                      
Finished  <<< planning_ros_msgs                                 [ 0.1 seconds ]
Finished  <<< rpg_single_board_io                               [ 0.1 seconds ]
Finished  <<< assimp_catkin                                     [ 0.1 seconds ]
Finished  <<< minimum_jerk_trajectories                         [ 0.1 seconds ]
Finished  <<< flightrender                                      [ 0.1 seconds ]
Finished  <<< quadrotor_msgs                                    [ 0.1 seconds ]
Starting  >>> agile_autonomy_msgs                                              
[... 90 lines cut ...]
    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /home/pc205/agile_autonomy_ws/catkin_aa/devel/share/open3d_conversions/cmake/open3d_conversionsConfig.cmake:173 (message):
  Project 'mpl_test_node' tried to find library 'Open3D'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'open3d_conversions'? Did you find_package() it before the subdirectory
  containing its code is included?
Call Stack (most recent call first):
  CMakeLists.txt:27 (find_package)

cd /home/pc205/agile_autonomy_ws/catkin_aa/build/mpl_test_node; catkin build --get-env mpl_test_node | catkin env -si  /usr/bin/cmake /home/pc205/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pc205/agile_autonomy_ws/catkin_aa/devel -DCMAKE_INSTALL_PREFIX=/home/pc205/agile_autonomy_ws/catkin_aa/install -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3; cd -

...............................................................................
Failed     << mpl_test_node:cmake                               [ Exited with code 1 ]
Failed    <<< mpl_test_node                                     [ 0.4 seconds ]
[build] Summary: 70 of 71 packages succeeded.                                  
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4.2 seconds total.

Howardcl, 2021-12-13:
[image]

racheraven, 2021-12-13:
Seems to be an open3d library error.
You may refer to [my answer ](https://github.com/uzh-rpg/agile_autonomy/issues/12#issuecomment-972612526) or [den250400's answer](https://github.com/uzh-rpg/agile_autonomy/issues/10#issuecomment-981095386) for possible solution.

Howardcl, 2021-12-13:
@racheraven Thank you for your reply! I have tried this method,but it don't works~

lilrookie76, 2021-12-25:
> 

Hey guy, have you solved this problem?

Howardcl, 2021-12-27:
@lilrookie76 yes! you can refer to https://blog.csdn.net/wxm__/article/details/121220816#t16

Howardcl, 2022-03-18:
done

closed 2022-03-18 by Howardcl