uzh-rpg/agile_autonomy #34: RuntimeError: Multiple packages found with the same name "example_control"

LiangtaoZhong, 2021-12-29:
When I run "catkin build", I received this error:

Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/zlt/agile_autonomy_ws/catkin_aa
--------------------------------------------------------------------------------
Build Space:        [exists] /home/zlt/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /home/zlt/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /home/zlt/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /home/zlt/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /home/zlt/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/bin/catkin", line 11, in <module>
    load_entry_point('catkin-tools==0.6.1', 'console_scripts', 'catkin')()
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 272, in main
    catkin_main(sysargs)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 267, in catkin_main
    sys.exit(args.main(args) or 0)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/cli.py", line 422, in main
    summarize_build=opts.summarize  # Can be True, False, or None
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 283, in build_isolated_workspace
    workspace_packages = find_packages(context.source_space_abs, exclude_subspaces=True, warnings=[])
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 96, in find_packages
    raise RuntimeError('\n'.join(duplicates))
**RuntimeError: Multiple packages found with the same name "example_control":**
- flightmare_agile_autonomy/examples/example_control
- rpg_flightmare/examples/example_control
Multiple packages found with the same name "example_pcd":
- flightmare_agile_autonomy/examples/example_pcd
- rpg_flightmare/examples/example_pcd
Multiple packages found with the same name "example_vison":
- flightmare_agile_autonomy/examples/example_vision
- rpg_flightmare/examples/example_vision
Multiple packages found with the same name "flightrender":
- flightmare_agile_autonomy/flightrender
- rpg_flightmare/flightrender
Multiple packages found with the same name "rpgq_command_bridge":
- flightmare_agile_autonomy/rpgq_command_bridge
- rpg_flightmare/rpgq_command_bridge
Multiple packages found with the same name "rpgq_common":
- flightmare_agile_autonomy/rpgq_common
- rpg_flightmare/rpgq_common
Multiple packages found with the same name "rpgq_components":
- flightmare_agile_autonomy/rpgq_components
- rpg_flightmare/rpgq_components
Multiple packages found with the same name "rpgq_msgs":
- flightmare_agile_autonomy/rpgq_msgs
- rpg_flightmare/rpgq_msgs
Multiple packages found with the same name "rpgq_simulator":
- flightmare_agile_autonomy/rpgq_simulator
- rpg_flightmare/rpgq_simulator

LiangtaoZhong, 2021-12-29:
I sloved it!

closed 2021-12-29 by LiangtaoZhong

Tian0906, 2023-08-11:
> I sloved it!

Hi, I have the same problem, how did you solve it?