uzh-rpg/agile_autonomy #36: Flightmare crashes on roslaunch agile_autonomy simulation.launch in docker

adam-mirza, 2022-01-20:
I have gotten all of the packages to build in an Ubuntu 20.04 docker container. When I launch agile_autonomy flightmare starts for about a second, then crashes.

Has anyone else experienced this issue, or have ideas on how to proceed?

I have attached the relevant terminal output below.

[image]

```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Ready ? : 0                                                         
[ INFO] [1642693599.801859432]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1642693599.803162953]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Ready ? : 0                                                         
RosPluginProvider.load(qt_gui_cpp/CppPluginProvider) exception raised in __builtin__.__import__(qt_gui_cpp.cpp_plugin_provider, [CppPlug
inProvider]):                                                       
Traceback (most recent call last): 
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_plugin_provider.py", line 33, in <module>
    from .cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name

RecursivePluginProvider.discover() loading plugin "qt_gui_cpp/CppPluginProvider" failed:
Traceback (most recent call last): 
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/recursive_plugin_provider.py", line 60, in discover
    instance = self._plugin_provider.load(plugin_descriptor.plugin_id(), None)
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 90, in load
    raise e
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_plugin_provider.py", line 33, in <module>
    from .cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name

Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
[flight_render-20] process has died [pid 654, exit code -11, cmd /home/usr/agile_autonomy_ws/catkin_aa/src/rpg_flightmare/flightrender/f
lightmare.x86_64 __name:=flight_render __log:=/root/.ros/log/278c32a4-7a08-11ec-a9b3-0242ac110002/flight_render-20.log].
log file: /root/.ros/log/278c32a4-7a08-11ec-a9b3-0242ac110002/flight_render-20*.log
```

psun-autel, 2022-04-10:
This has something to do with your PyQt related installation. I ran into a couple of issues, and this is one of them. I basically did qt related libraries installation over and over, and made sure I didn't mess it with any conda environment. At least, I got over this problem by now.

mw9385, 2022-05-23:
DId you solve the issue?

tiankegangHW, 2022-09-06:
hello, I also build all of the packages in an Ubuntu 20.04 docker container. The container has X11 forwarding setup (on /tmp/.X11-unix) and is run with the --privileged flag. However, the environment can not be loaded and flightmare can not run. The error is bellow
[image]
The standalone.tar is extracted in "...rpg_flightmare/flightrender" and  $RPGQ_PARAM_DIR is set correctly. Rviz can also work properly.
Thank you very much!!

mw9385, 2022-09-06:
@tiankegangHW Hi, when you running it on docker, it doesn't work. Try turning on simulator and then run the launch file.

AnshShah3009, 2023-10-11:
> Did you solve the issue? I faced with exact the same problem.

Did you solve this finally?