uzh-rpg/agile_autonomy #37: Error while 'catkin build'

Kevinlibaba, 2022-01-23:
**Environment**: ubuntu 18.04, ROS Melodic, cmake version 3.22.1, gcc/g++7.5. open3d v0.14 installed.

The following errors occurred when I tried `catkin build` step.

I have tried to set `set(CMAKE_CXX_FLAGS "-std=c++14 ")` in CMakeList.txt, but it didn't work for that error.

Thanks for any helps!

```
Errors     << mpl_test_node:make /home/rui/agile_autonomy_ws/catkin_aa/logs/mpl_test_node/build.make.001.log
In file included from /usr/local/include/open3d/geometry/PointCloud.h:36:0,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/ellipsoid_util.h:11,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/env_cloud.h:8,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/ellipsoid_planner.h:9,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:3:
/usr/local/include/open3d/utility/Optional.h:308:18: error: ‘constexpr T& open3d::utility::optional<T>::contained_val() const &’ cannot be overloaded
     constexpr T& contained_val() & { return OptionalBase<T>::storage_.value_; }
                  ^~~~~~~~~~~~~
/usr/local/include/open3d/utility/Optional.h:302:24: error: with ‘constexpr const T& open3d::utility::optional<T>::contained_val() const &’
     constexpr const T& contained_val() const& {
                        ^~~~~~~~~~~~~
/usr/local/include/open3d/utility/Optional.h:478:36: error: ‘constexpr T* open3d::utility::optional<T>::operator->() const’ cannot be overloaded
     TR2_OPTIONAL_HOST_CONSTEXPR T* operator->() {
                                    ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:474:42: error: with ‘constexpr const T* open3d::utility::optional<T>::operator->() const’
     TR2_OPTIONAL_HOST_CONSTEXPR T const* operator->() const {
                                          ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:487:36: error: ‘constexpr T& open3d::utility::optional<T>::operator*() const &’ cannot be overloaded
     TR2_OPTIONAL_HOST_CONSTEXPR T& operator*() & {
                                    ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:483:42: error: with ‘constexpr const T& open3d::utility::optional<T>::operator*() const &’
     TR2_OPTIONAL_HOST_CONSTEXPR T const& operator*() const& {
                                          ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:504:36: error: ‘constexpr T& open3d::utility::optional<T>::value() const &’ cannot be overloaded
     TR2_OPTIONAL_HOST_CONSTEXPR T& value() & {
                                    ^~~~~
/usr/local/include/open3d/utility/Optional.h:497:42: error: with ‘constexpr const T& open3d::utility::optional<T>::value() const &’
     TR2_OPTIONAL_HOST_CONSTEXPR T const& value() const& {
                                          ^~~~~
In file included from /usr/local/include/open3d/core/Tensor.h:42:0,
                 from /usr/local/include/open3d/core/EigenConverter.h:34,
                 from /usr/local/include/open3d/Open3D.h:38,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:19:
/usr/local/include/open3d/core/TensorInit.h:95:49: error: ‘std::index_sequence’ has not been declared
 SizeVector InitializerShape(const L& list, std::index_sequence<D...>) {
                                                 ^~~~~~~~~~~~~~
/usr/local/include/open3d/core/TensorInit.h:95:63: error: expected ‘,’ or ‘...’ before ‘<’ token
 SizeVector InitializerShape(const L& list, std::index_sequence<D...>) {
                                                               ^
/usr/local/include/open3d/core/TensorInit.h: In function ‘open3d::core::SizeVector open3d::core::tensor_init::InferShape(const L&)’:
/usr/local/include/open3d/core/TensorInit.h:103:24: error: ‘make_index_sequence’ is not a member of ‘std’
             list, std::make_index_sequence<InitializerDim<L>::value>());
                        ^~~~~~~~~~~~~~~~~~~
/usr/local/include/open3d/core/TensorInit.h:103:24: note: suggested alternative:
In file included from /usr/local/include/open3d/utility/Logging.h:47:0,
                 from /usr/local/include/open3d/core/Device.h:32,
                 from /usr/local/include/open3d/core/Blob.h:33,
                 from /usr/local/include/open3d/Open3D.h:34,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:19:
/usr/local/include/open3d/3rdparty/fmt/ranges.h:141:66: note:   ‘fmt::v6::internal::make_index_sequence’
 using make_index_sequence = make_integer_sequence<std::size_t, N>;
                                                                  ^
In file included from /usr/local/include/open3d/core/Tensor.h:42:0,
                 from /usr/local/include/open3d/core/EigenConverter.h:34,
                 from /usr/local/include/open3d/Open3D.h:38,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:19:
/usr/local/include/open3d/core/TensorInit.h:103:70: error: expected primary-expression before ‘)’ token
             list, std::make_index_sequence<InitializerDim<L>::value>());
                                                                      ^
In file included from /usr/local/include/open3d/visualization/gui/UIImage.h:30:0,
                 from /usr/local/include/open3d/visualization/gui/Button.h:33,
                 from /usr/local/include/open3d/Open3D.h:119,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
                 from /home/rui/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:19:
/usr/local/include/open3d/visualization/rendering/RendererHandle.h: At global scope:
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:189:35: error: ‘format’ function uses ‘auto’ type specifier without trailing return type
                 FormatContext& ctx) {
                                   ^
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:189:35: note: deduced return type only available with -std=c++14 or -std=gnu++14
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:197:43: error: ‘parse’ function uses ‘auto’ type specifier without trailing return type
     constexpr auto parse(ParseContext& ctx) {
                                           ^
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:197:43: note: deduced return type only available with -std=c++14 or -std=gnu++14
make[2]: *** [CMakeFiles/ellipsoid_planner_node.dir/src/ellipsoid_planner_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/ellipsoid_planner_node.dir/all] Error 2
make: *** [all] Error 2
```

PengqinWANG, 2022-02-11:
Hi Kevin,
Have you solved the problem? I am faced with the same one.

Kevinlibaba, 2022-02-11:
@PengqinWANG I have tried Open3D v0.9.0, anything went well. You can just follow this [comments](https://github.com/uzh-rpg/agile_autonomy/issues/10#issuecomment-981095386).

closed 2022-02-11 by Kevinlibaba

PengqinWANG, 2022-02-11:
Thank you so much! It really works!

PengqinWANG, 2022-02-12:
Hi Kevin, have you met this problem?
/home/pengqin/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/src/agile_autonomy.cpp:176:22: error: ‘class fpv_aggressive_trajectories::AcrobaticSequence’ has no member named ‘appendStraight’
   acrobatic_sequence.appendStraight(end_position_1, end_velocity_1, end_yaw
I found [acrobatic_sequence.h](https://github.com/uzh-rpg/deep_drone_acrobatics/blob/dependabot/pip/tensorflow-gpu-2.5.3/acrobatic_trajectory_helper/include/acrobatic_trajectory_helper/acrobatic_sequence.h) in the repository [deep_drone_acrobatics](https://github.com/uzh-rpg/deep_drone_acrobatics).
But what confused me is that there is no defination of appendStraight. So tricky

Kevinlibaba, 2022-02-13:
Sorry Wang, I havn't met this problem.
> Hi Kevin, have you met this problem?
> /home/pengqin/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/src/agile_autonomy.cpp:176:22: error: ‘class [...]