uzh-rpg/agile_autonomy #43: Error occurs when catkin build `mpl_test_node`

zhengzaiyi, 2022-03-17:
I was running `catkin b` and some strange events happened on `mpl_test_node`. As you can see in the following error message, in my opinion it seems like something to do with C/C++ compiler. The versions of gcc/g++ are both 7.5.0 
```
Errors     << mpl_test_node:make /home/m/agile_autonomy_ws/catkin_aa/logs/mpl_test_node/build.make.010.log
In file included from /usr/local/include/open3d/geometry/PointCloud.h:36:0,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/ellipsoid_util.h:11,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/env_cloud.h:8,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/ellipsoid_planner.h:9,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:3:
/usr/local/include/open3d/utility/Optional.h:308:18: error: ‘constexpr T& open3d::utility::optional<T>::contained_val() const &’ cannot be overloaded
     constexpr T& contained_val() & { return OptionalBase<T>::storage_.value_; }
                  ^~~~~~~~~~~~~
/usr/local/include/open3d/utility/Optional.h:302:24: error: with ‘constexpr const T& open3d::utility::optional<T>::contained_val() const &’
     constexpr const T& contained_val() const& {
                        ^~~~~~~~~~~~~
/usr/local/include/open3d/utility/Optional.h:478:36: error: ‘constexpr T* open3d::utility::optional<T>::operator->() const’ cannot be overloaded
     TR2_OPTIONAL_HOST_CONSTEXPR T* operator->() {
                                    ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:474:42: error: with ‘constexpr const T* open3d::utility::optional<T>::operator->() const’
     TR2_OPTIONAL_HOST_CONSTEXPR T const* operator->() const {
                                          ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:487:36: error: ‘constexpr T& open3d::utility::optional<T>::operator*() const &’ cannot be overloaded
     TR2_OPTIONAL_HOST_CONSTEXPR T& operator*() & {
                                    ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:483:42: error: with ‘constexpr const T& open3d::utility::optional<T>::operator*() const &’
     TR2_OPTIONAL_HOST_CONSTEXPR T const& operator*() const& {
                                          ^~~~~~~~
/usr/local/include/open3d/utility/Optional.h:504:36: error: ‘constexpr T& open3d::utility::optional<T>::value() const &’ cannot be overloaded
     TR2_OPTIONAL_HOST_CONSTEXPR T& value() & {
                                    ^~~~~
/usr/local/include/open3d/utility/Optional.h:497:42: error: with ‘constexpr const T& open3d::utility::optional<T>::value() const &’
     TR2_OPTIONAL_HOST_CONSTEXPR T const& value() const& {
                                          ^~~~~
In file included from /usr/local/include/open3d/core/Tensor.h:43:0,
                 from /usr/local/include/open3d/core/EigenConverter.h:34,
                 from /usr/local/include/open3d/Open3D.h:38,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:19:
/usr/local/include/open3d/core/TensorInit.h:95:49: error: ‘std::index_sequence’ has not been declared
 SizeVector InitializerShape(const L& list, std::index_sequence<D...>) {
                                                 ^~~~~~~~~~~~~~
/usr/local/include/open3d/core/TensorInit.h:95:63: error: expected ‘,’ or ‘...’ before ‘<’ token
 SizeVector InitializerShape(const L& list, std::index_sequence<D...>) {
                                                               ^
/usr/local/include/open3d/core/TensorInit.h: In function ‘open3d::core::SizeVector open3d::core::tensor_init::InferShape(const L&)’:
/usr/local/include/open3d/core/TensorInit.h:103:24: error: ‘make_index_sequence’ is not a member of ‘std’
             list, std::make_index_sequence<InitializerDim<L>::value>());
                        ^~~~~~~~~~~~~~~~~~~
/usr/local/include/open3d/core/TensorInit.h:103:24: note: suggested alternatives:
In file included from /usr/include/boost/mp11/algorithm.hpp:25:0,
                 from /usr/include/boost/mp11/bind.hpp:11,
                 from /usr/include/boost/parameter/aux_/is_placeholder.hpp:46,
                 from /usr/include/boost/parameter/value_type.hpp:101,
                 from /usr/include/boost/parameter/aux_/name.hpp:31,
                 from /usr/include/boost/parameter/name.hpp:10,
                 from /usr/include/boost/heap/policies.hpp:13,
                 from /usr/include/boost/heap/detail/stable_heap.hpp:20,
                 from /usr/include/boost/heap/d_ary_heap.hpp:21,
                 from /home/m/agile_autonomy_ws/catkin_aa/devel/share/motion_primitive_library/cmake/../../../include/mpl_planner/common/state_space.h:10,
                 from /home/m/agile_autonomy_ws/catkin_aa/devel/share/motion_primitive_library/cmake/../../../include/mpl_planner/common/graph_search.h:10,
                 from /home/m/agile_autonomy_ws/catkin_aa/devel/share/motion_primitive_library/cmake/../../../include/mpl_planner/common/planner_base.h:12,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_external_planner/include/mpl_external_planner/ellipsoid_planner/ellipsoid_planner.h:10,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:3:
/usr/include/boost/mp11/integer_sequence.hpp:104:90: note:   ‘boost::mp11::make_index_sequence’
 template<std::size_t N> using make_index_sequence = make_integer_sequence<std::size_t, N>;
                                                                                          ^
In file included from /usr/local/include/open3d/utility/Logging.h:42:0,
                 from /usr/local/include/open3d/core/Device.h:32,
                 from /usr/local/include/open3d/core/Blob.h:33,
                 from /usr/local/include/open3d/Open3D.h:34,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
[... 10 lines cut ...]
             list, std::make_index_sequence<InitializerDim<L>::value>());
                                                                      ^
In file included from /usr/local/include/open3d/visualization/gui/UIImage.h:30:0,
                 from /usr/local/include/open3d/visualization/gui/Button.h:33,
                 from /usr/local/include/open3d/Open3D.h:119,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19,
                 from /home/m/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/mpl_test_node/src/ellipsoid_planner_node.cpp:19:
/usr/local/include/open3d/visualization/rendering/RendererHandle.h: At global scope:
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:181:35: error: ‘format’ function uses ‘auto’ type specifier without trailing return type
                 FormatContext& ctx) {
                                   ^
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:181:35: note: deduced return type only available with -std=c++14 or -std=gnu++14
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:189:43: error: ‘parse’ function uses ‘auto’ type specifier without trailing return type
     constexpr auto parse(ParseContext& ctx) {
                                           ^
/usr/local/include/open3d/visualization/rendering/RendererHandle.h:189:43: note: deduced return type only available with -std=c++14 or -std=gnu++14
make[2]: *** [CMakeFiles/ellipsoid_planner_node.dir/build.make:63：CMakeFiles/ellipsoid_planner_node.dir/src/ellipsoid_planner_node.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:4364：CMakeFiles/ellipsoid_planner_node.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
cd /home/m/agile_autonomy_ws/catkin_aa/build/mpl_test_node; catkin build --get-env mpl_test_node | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

....................................................................................................
Failed     << mpl_test_node:make                          [ Exited with code 2 ]                    
Failed    <<< mpl_test_node                               [ 27.0 seconds ]
```

halil93ibrahim, 2022-03-30:
For me, building Open3D v0.9.0 solves the errors for mpl_test_node.

zhengzaiyi, 2022-04-20:
> For me, building Open3D v0.9.0 solves the errors for mpl_test_node.

Yes, that's the right way.

closed 2022-04-20 by zhengzaiyi

stochasticritic, 2022-08-22:
I've got Open3D v0.9.0 installed (replaced the latest v.0.15.1 with this in my repo) and it has resolved my error with the _traj_sampler_ package, but this error is not yet resolved in my case. This is the only package that failing as of now. Looks like there are some cmake/C++ compilation errors.

Please help me in this case.

```
sohamkulkarni@sohamxlegion:~/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros$ catkin build
------------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/sohamkulkarni/agile_autonomy_ws/catkin_aa
------------------------------------------------------------------------------------
Build Space:        [exists] /home/sohamkulkarni/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /home/sohamkulkarni/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /home/sohamkulkarni/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /home/sohamkulkarni/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /home/sohamkulkarni/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
------------------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------------------
[build] Found 71 packages in 0.0 seconds.                                                                                                                                                                         
[build] Package table is up to date.                                                                                                                                                                              
Starting  >>> catkin_boost_python_buildtool                                                                                                                                                                       
Starting  >>> catkin_simple                                                                                                                                                                                       
Starting  >>> decomp_util                                                                                                                                                                                         
Starting  >>> mav_msgs                                                                                                                                                                                            
Starting  >>> mav_state_machine_msgs                                                                                                                                                                              
Starting  >>> mav_system_msgs                                                                                                                                                                                     
Starting  >>> motion_primitive_library                                                                                                                                                                            
Starting  >>> open3d_conversions                                                                                                                                                                                  
Starting  >>> rotors_comm                                                                                                                                                                                         
Starting  >>> rotors_description                                                                                                                                                                                  
Starting  >>> rotors_evaluation                                                                                                                                                                                   
Starting  >>> rqt_rotors                                                                                                                                                                                          
Finished  <<< catkin_simple                                     [ 0.1 seconds ]                                                                                                                                   
Starting  >>> assimp_catkin                                                                                                                                                                                       
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                           
[build] Runtime: 27.1 seconds total.                                                                                                                                                                              
sohamkulkarni@sohamxlegion:~/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros$ 

```

NIHALNAVALE, 2023-03-15:
> For me, building Open3D v0.9.0 solves the errors for mpl_test_node.

My current version is 0.13, how do I change it to 0.9?
Can you share the steps?