uzh-rpg/agile_autonomy #46: Cannot load simulation environment

zhengzaiyi, 2022-04-07:
When I was running the command `roslaunch agile_autonomy simulation.launch`, something wrong happened as the following image shows: 
[image]
It kept printing `Ready ? : 0`, the `standalone` has been extracted to the 'flightrender` folder. Is it necessary to build the `standalone` in advance?

psun-autel, 2022-04-10:
Waiting on further instruction as well. I don't have any errors in your snapshot anymore, but the other outputs are the same as mine. 
Also, I got further error 

 INFO] [1649622827.802552360, 2.876000000]: Unity is ready!
[ERROR] [1649622828.212042755, 3.080000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.321543409, 3.135000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.459425264, 3.203000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.582797969, 3.265000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.721430274, 3.334000000]: no kernel image is available for execution on the device 209

^C[ERROR] [1649622828.859060628, 3.403000000]: no kernel image is available for execution on the device 209

psun-autel, 2022-04-11:
You need to click start and also click arm bridge to see the effect.

zhengzaiyi, 2022-04-20:
> 

If you're using GPU with Nvidia Ampere structure, you need to add the gpu-code in the CmakeLists

zhengzaiyi, 2022-04-20:
> Waiting on further instruction as well. I don't have any errors in your snapshot anymore, but the other outputs are the same as mine. Also, I got further error
>
> INFO] [1649622827.802552360, [...]

Maybe you are using Nvidia 30XX? If your Gpu is of Ampere structure, you need to add the corresponding gpu-code to CmakeLists.

mw9385, 2022-05-20:
Have you solved this issue?

Hang-Shi, 2022-06-19:
> Maybe you are using Nvidia 30XX? If your Gpu is of Ampere structure, you need to add the corresponding gpu-code to CmakeLists.

Buddy, could you tell me in detail how to add the corresponding gpu-code to CmakeLists. 1) I'm not sure how to determine the corresponding gpu-code. 2) There are so many CmakeLists that I don't know which one is suitable. 
Waiting for your reply.    ^_^

Hang-Shi, 2022-06-19:
> Maybe you are using Nvidia 30XX? If your Gpu is of Ampere structure, you need to add the corresponding gpu-code to CmakeLists.

I solved the problem. 1) find the path   /agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/CmakeLists.    2)Open the CmakeLists and add  '-gencode=arch=compute_80,code=sm_80'  and  '-gencode=arch=compute_86,code=sm_86'

Hang-Shi, 2022-06-19:
> Did you solve this issue?

Now you can solve this issue

zhengzaiyi, 2022-07-06:
> You need to click start and also click arm bridge to see the effect.

HangShi's right, it's all about the gpu structure(I guess you're using Turing)

stary-T-sky, 2023-01-04:
Did you solve this issue? My terminal also kept printing "Ready ? : 0".

zhengzaiyi, 2023-01-05:
> Did you solve this issue? My terminal also kept printing "Ready ? : 0".

Yeah, see [this quote](https://github.com/uzh-rpg/agile_autonomy/issues/46#issuecomment-1159749990)

closed 2023-01-05 by zhengzaiyi

9woods123, 2024-03-20:
> Did you solve this issue? My terminal also kept printing "Ready ? : 0".
Should I download the unity exe, and put it in this path "~/agile_autonomy_ws/catkin_aa/src/rpg_flightmare/flightrender" ? This operation is not noted in tutorial or wiki.

9woods123, 2024-03-20:
> > Did you solve this issue? My terminal also kept printing "Ready ? : 0".
> 
> Yeah, see [this quote](https://github.com/uzh-rpg/agile_autonomy/issues/46#issuecomment-1159749990)

Do I need to download an unity exe ,and put it in "~/agile_autonomy_ws/catkin_aa/src/rpg_flightmare/flightrender" ?
This operation is not mentioned in tutorials or wiki, and it confuses me .