uzh-rpg/agile_autonomy #47: Further instruction needed for someone new to this.

psun-autel, 2022-04-10:
After running roslaunch agile_autonomy simulation.launch

I am not sure what I should expect or what I should do next in order to see this demo. In the Unity window, the environment is being rendered in real time, and you can observe the grass is moving. The view angle is fixed. 

The partial console output is shown as below. 

roslaunch agile_autonomy simulation.launch

[ INFO] [1649622819.898319698]: [/hummingbird/agile_autonomy] camera/baseline = 0.1
[ INFO] [1649622819.898993112]: [/hummingbird/agile_autonomy] camera/pitch_angle_deg = 0
Vertical Fov is 74.7021
Initializing ZMQ connections...
Initializing ZMQ done...
[ INFO] [1649622819.901140818]: Connecting to unity...
Ready ? : 0
[ INFO] [1649622820.083847360]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1649622820.084971604]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
[ INFO] [1649622821.892760684]: waitForService: Service [/gazebo/set_physics_properties] is now available.
Ready ? : 0
[ INFO] [1649622821.910235945]: Physics dynamic reconfigure ready.
[INFO] [1649622821.992587, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1649622822.084647, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
[hummingbird/spawn_hummingbird-3] process has finished cleanly
log file: /home/psun/.ros/log/808eb694-b90d-11ec-b32c-11a93648a3ed/hummingbird-spawn_hummingbird-3*.log
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 1
[ INFO] [1649622827.802552360, 2.876000000]: Unity is ready!
[ERROR] [1649622828.212042755, 3.080000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.321543409, 3.135000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.459425264, 3.203000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.582797969, 3.265000000]: no kernel image is available for execution on the device 209

[ERROR] [1649622828.721430274, 3.334000000]: no kernel image is available for execution on the device 209

^C[ERROR] [1649622828.859060628, 3.403000000]: no kernel image is available for execution on the device 209

psun-autel, 2022-04-11:
I figured out myself. Closing this one.

closed 2022-04-11 by psun-autel

blu666, 2022-12-29:
How did you fix this? I'm running into the same problem. Would be great if you could share your solution!! Thank you!

blu666, 2023-01-02:
I resolved the "no kernel image is available" issue by following #18 and editing line 14 in the agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt  according to GPU architecture

random-user-in-space, 2023-01-05:
Can you tell me what did you use? I am using NVIDIA GeForce RTX 3060 Laptop GPU with cuda 11.2

blu666, 2023-01-05:
@random-user-in-space I am using NVIDIA RTX A2000 ti laptop GPU which I believe is equivalent to 3050. I also installed cuda 11.2. After changing the CMakeList, sometimes it will throw a lot of warning messages about sm_86, but according to other sources these messages can be ignored. Hope this helps!

random-user-in-space, 2023-01-06:
@blu666 Thank you for quick response. Do you mind sharing the exact config that you used?

I used the below config but I get build error if I use that. 
set(CUDA_NVCC_FLAGS
        ${CUDA_NVCC_FLAGS};
        -O3
        -gencode=arch=compute_52,code=sm_52
        -gencode=arch=compute_61,code=sm_61
        -gencode=arch=compute_62,code=sm_62
        -gencode=arch=compute_75,code=sm_75
        -gencode=arch=compute_86,code=sm_86

        )

error while building-->

Errors     << agile_autonomy:make /home/sumit/agile_autonomy_ws/catkin_aa/logs/agile_autonomy/build.make.041.log
nvcc fatal   : Unsupported gpu architecture 'compute_86'
CMake Error at sgm_gpu_generated_median_filter.cu.o.Release.cmake:220 (message):
  Error generating
  /home/sumit/agile_autonomy_ws/catkin_aa/build/agile_autonomy/CMakeFiles/sgm_gpu.dir/src/sgm_gpu/./sgm_gpu_generated_median_filter.cu.o

make[2]: *** [CMakeFiles/sgm_gpu.dir/build.make:2289: CMakeFiles/sgm_gpu.dir/src/sgm_gpu/sgm_gpu_generated_median_filter.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
nvcc fatal   : Unsupported gpu architecture 'compute_86'
CMake Error at sgm_gpu_generated_costs.cu.o.Release.cmake:220 (message):
  Error generating
  /home/sumit/agile_autonomy_ws/catkin_aa/build/agile_autonomy/CMakeFiles/sgm_gpu.dir/src/sgm_gpu/./sgm_gpu_generated_costs.cu.o

nvcc fatal   : Unsupported gpu architecture 'compute_86'
CMake Error at sgm_gpu_generated_sgm_gpu.cu.o.Release.cmake:220 (message):
  Error generating
  /home/sumit/agile_autonomy_ws/catkin_aa/build/agile_autonomy/CMakeFiles/sgm_gpu.dir/src/sgm_gpu/./sgm_gpu_generated_sgm_gpu.cu.o

make[2]: *** [CMakeFiles/sgm_gpu.dir/build.make:1673: CMakeFiles/sgm_gpu.dir/src/sgm_gpu/sgm_gpu_generated_costs.cu.o] Error 1
make[2]: *** [CMakeFiles/sgm_gpu.dir/build.make:1498: CMakeFiles/sgm_gpu.dir/src/sgm_gpu/sgm_gpu_generated_sgm_gpu.cu.o] Error 1
nvcc fatal   : Unsupported gpu architecture 'compute_86'
CMake Error at sgm_gpu_generated_hamming_cost.cu.o.Release.cmake:220 (message):
  Error generating
  /home/sumit/agile_autonomy_ws/catkin_aa/build/agile_autonomy/CMakeFiles/sgm_gpu.dir/src/sgm_gpu/./sgm_gpu_generated_hamming_cost.cu.o

nvcc fatal   : Unsupported gpu architecture 'compute_86'
CMake Error at sgm_gpu_generated_left_right_consistency.cu.o.Release.cmake:220 (message):
  Error generating
  /home/sumit/agile_autonomy_ws/catkin_aa/build/agile_autonomy/CMakeFiles/sgm_gpu.dir/src/sgm_gpu/./sgm_gpu_generated_left_right_consistency.cu.o

make[2]: *** [CMakeFiles/sgm_gpu.dir/build.make:1940: CMakeFiles/sgm_gpu.dir/src/sgm_gpu/sgm_gpu_generated_hamming_cost.cu.o] Error 1
make[2]: *** [CMakeFiles/sgm_gpu.dir/build.make:2115: CMakeFiles/sgm_gpu.dir/src/sgm_gpu/sgm_gpu_generated_left_right_consistency.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1557: CMakeFiles/sgm_gpu.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

blu666, 2023-01-26:
@random-user-in-space Have you resolved this issue?

I've encountered this error before, but it went away. It is a weird error. 

"nvcc fatal : Unsupported gpu architecture 'compute_86'" but 'compute_86' is a allowed value according to the last line. Try restarting the computer (: Or might be related to something in your Cuda and driver.