uzh-rpg/agile_autonomy #48: can't fly when run test_trajectories.py

AndyYan2, 2022-04-15:
When run "python test_trajectories.py --settings_file=config/test_settings.yaml", it will loop as follows:
`==========================
     RESET SIMULATION     
==========================

Unpausing Physics...

Placing quadrotor...
success: True
status_message: "SetModelState: set model state done"
Received call to Clear Buffer and Restart Experiment

Resetting experiment

Done Reset
Doing experiment 0
Current experiment failed. Will try again
==========================
     RESET SIMULATION     
==========================

Unpausing Physics...

Placing quadrotor...
success: True
status_message: "SetModelState: set model state done"
Received call to Clear Buffer and Restart Experiment

Resetting experiment

Done Reset
Doing experiment 0
Current experiment failed. Will try again
`
In rviz, the RGM is NO Image
[image]
Check the rostopic, I find many topics received no message. 
[image]
What's wrong with this? It's the fightmare can't  communicate with ROS ? Thanks a lot !

AndyYan2, 2022-04-20:
Maybe it's because the GPU driver version mismatches since the image can't show. I have solved the problem by reinstall the operating system.

closed 2022-04-20 by AndyYan2

mw9385, 2022-05-24:
@AndyYan2 Hi, could you write down your hardware specification (including GPU, GPU driver version, cuda version)?
I am struggling with similar issue. Thanks in advance :)

AndyYan2, 2022-05-24:
My hardware version is as follows:
ubuntu :18.04
CUDA :11.0
GPU :NVIDIA GeForce RTX 2060
GPU driver : 450.51.05
I installed the driver by CUDA. When installing CUDA, it will install the driver automatically.
gcc :7.5.0
g++ :7.5.0
python :3.9.7
open3d : 0.9.0 
Hope it's useful !