uzh-rpg/agile_autonomy #52: Flightmare stops when roslaunch agile_autonomy simulation.launch

tmddn833, 2022-05-17:
Hello! I have a trouble with Flightmare simulator.
Here is my roslaunch agile_autonomy simulation.launch result
[simulation_log.txt](https://github.com/uzh-rpg/agile_autonomy/files/8704766/simulation_log.txt)
As far as I know, this message have no error or ploblem to excute the all nodes in launch file.
```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/home/seungwooubuntu/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/launch/simulation.launch
[93mredefining global symbol: pi[0m
when processing file: /home/seungwooubuntu/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/resources/urdf/component_snippets.xacro
included from: /home/seungwooubuntu/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/resources/urdf/hummingbird_base.xacro
included from: /home/seungwooubuntu/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/resources/urdf/mav_generic_odometry_sensor_rgb.gazebo
[1mstarted roslaunch server http://seungwooubuntu:41435/[0m

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /hummingbird/agile_autonomy/Q_attitude: 200
 * /hummingbird/agile_autonomy/Q_perception: 0
 * /hummingbird/agile_autonomy/Q_pos_xy: 100
 * /hummingbird/agile_autonomy/Q_pos_z: 100
 * /hummingbird/agile_autonomy/Q_velocity: 10
 * /hummingbird/agile_autonomy/R_pitchroll: 1.0
 * /hummingbird/agile_autonomy/R_thrust: 1.0
 * /hummingbird/agile_autonomy/R_yaw: 1.0
 * /hummingbird/agile_autonomy/camera/baseline: 0.1
 * /hummingbird/agile_autonomy/camera/fov: 91.0
 * /hummingbird/agile_autonomy/camera/height: 480
 * /hummingbird/agile_autonomy/camera/pitch_angle_deg: 0.0
 * /hummingbird/agile_autonomy/camera/width: 640
 * /hummingbird/agile_autonomy/data_dir: /home/seungwooubu...
 * /hummingbird/agile_autonomy/data_generation/save_freq: 15.0
 * /hummingbird/agile_autonomy/general/control_command_delay: 0.02
 * /hummingbird/agile_autonomy/general/num_traj_viz: 1000
 * /hummingbird/agile_autonomy/general/nw_predictions_frame_id: 1
 * /hummingbird/agile_autonomy/general/perform_global_planning: False
 * /hummingbird/agile_autonomy/general/process_every_nth_odometry: 20
 * /hummingbird/agile_autonomy/general/real_world_exp: False
 * /hummingbird/agile_autonomy/general/test_time_max_z: 30.0
 * /hummingbird/agile_autonomy/general/test_time_min_z: 1.0
 * /hummingbird/agile_autonomy/general/test_time_velocity: 7.0
......
(skip)
......
[0m[ INFO] [1652756385.966290367]: [/hummingbird/agile_autonomy] camera/height = 480[0m
[0m[ INFO] [1652756385.966524840]: [/hummingbird/agile_autonomy] camera/baseline = 0.1[0m
[0m[ INFO] [1652756385.966882249]: [/hummingbird/agile_autonomy] camera/pitch_angle_deg = 0[0m
Vertical Fov is 74.7021
Initializing ZMQ connections...
Initializing ZMQ done...
[0m[ INFO] [1652756385.968852129]: Connecting to unity...[0m
Ready ? : 0
[INFO] [1652756385.995516, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1652756385.997963, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
Ready ? : 0
[0m[ INFO] [1652756386.242429703]: Finished loading Gazebo ROS API Plugin.[0m
[0m[ INFO] [1652756386.243175695]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...[0m
Ready ? : 0
Ready ? : 0
[0m[ INFO] [1652756386.572179509, 0.005000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.[0m
[0m[ INFO] [1652756386.578769584, 0.008000000]: Physics dynamic reconfigure ready.[0m
[INFO] [1652756386.601708, 0.000000]: Calling service /gazebo/spawn_urdf_model
Ready ? : 0
[INFO] [1652756386.792584, 0.091000]: Spawn status: SpawnModel: Successfully spawned entity
Ready ? : 0
Ready ? : 0
Ready ? : 0
[1m[hummingbird/spawn_hummingbird-3] process has finished cleanly
log file: /home/seungwooubuntu/.ros/log/67b1578e-d58d-11ec-9ee8-ffb260b24679/hummingbird-spawn_hummingbird-3*.log[0m
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 1
[0m[ INFO] [1652756393.909530532, 3.647000000]: Unity is ready![0m
```
**With the message "Unity is ready!", the flightmare change to other envirenment(like desert) and stop.**
After a few second, Flightmare shows this message
[image]

I installed Open3D(v0.9.0) and all ros packages following your step by step instruction. and I built every packages 
I also installed.

My cuda version is 11.2, nvidia gtx 2060, 32G RAM, intel i7-12700 cpu.
I don't thins this is because of computer spec.

Please check my issue and reply.
Thank you!

mw9385, 2022-05-20:
Faced with same issue!

tmddn833, 2022-05-20:
I think it solved now...
In ~/.bashrc, set the directory of cuda path like this
```
export PATH="/usr/local/cuda-<your cuda version>/bin:$PATH"
```
after set cuda path clean the build history and rebuild it.
```
$ cd agile_autonomy_ws/catkin_aa/
$ catkin clean
<type y for question>
$ catkin build
```
Before this debugging, it runs very rarely but now it runs all time i execute.
**But it works not smooth, and there is no drone rendering in the flightmare window.**
Does this right execution??

mw9385, 2022-05-20:
> I think it solved now... In ~/.bashrc, set the directory of cuda path like this
> 
> ```
> export PATH="/usr/local/cuda-<your cuda version>/bin:$PATH"
> ```
> 
> after set cuda path clean the build history and rebuild it.
> 
> ```
> $ cd agile_autonomy_ws/catkin_aa/
> $ catkin clean
> <type y for question>
> $ catkin build
> ```
> 
> Before this debugging, it runs very rarely but now it runs all time i execute. **But it works not smooth, and there is no drone rendering in the flightmare window.** Is there any problem for this??

I have tried this method, but I still encountered the same problem. After launching the `simulation.launch`, The Ready continuously shows 0 and I could not see any images from fligtmare simulator. 

```root@unist-System-Product-Name:~/agile_autonomy_ws# roslaunch agile_autonomy simulation.launch
... logging to /root/.ros/log/749f3594-d815-11ec-803e-88366cf6d71f/roslaunch-unist-System-Product-Name-44991.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global symbol: pi
when processing file: /root/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/resources/urdf/component_snippets.xacro
included from: /root/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/resources/urdf/hummingbird_base.xacro
included from: /root/agile_autonomy_ws/catkin_aa/src/agile_autonomy/data_generation/agile_autonomy/resources/urdf/mav_generic_odometry_sensor_rgb.gazebo
started roslaunch server http://unist-System-Product-Name:33939/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /hummingbird/agile_autonomy/Q_attitude: 200
 * /hummingbird/agile_autonomy/Q_perception: 0
 * /hummingbird/agile_autonomy/Q_pos_xy: 100
 * /hummingbird/agile_autonomy/Q_pos_z: 100
 * /hummingbird/agile_autonomy/Q_velocity: 10
 * /hummingbird/agile_autonomy/R_pitchroll: 1.0
 * /hummingbird/agile_autonomy/R_thrust: 1.0
 * /hummingbird/agile_autonomy/R_yaw: 1.0
 * /hummingbird/agile_autonomy/camera/baseline: 0.1
 * /hummingbird/agile_autonomy/camera/fov: 91.0
 * /hummingbird/agile_autonomy/camera/height: 480
 * /hummingbird/agile_autonomy/camera/pitch_angle_deg: 0.0
 * /hummingbird/agile_autonomy/camera/width: 640
 * /hummingbird/agile_autonomy/data_dir: /root/agile_auton...
 * /hummingbird/agile_autonomy/data_generation/save_freq: 15.0
 * /hummingbird/agile_autonomy/general/control_command_delay: 0.02
 * /hummingbird/agile_autonomy/general/num_traj_viz: 1000
 * /hummingbird/agile_autonomy/general/nw_predictions_frame_id: 1
 * /hummingbird/agile_autonomy/general/perform_global_planning: False
 * /hummingbird/agile_autonomy/general/process_every_nth_odometry: 20
 * /hummingbird/agile_autonomy/general/real_world_exp: False
 * /hummingbird/agile_autonomy/general/test_time_max_z: 30.0
 * /hummingbird/agile_autonomy/general/test_time_min_z: 1.0
 * /hummingbird/agile_autonomy/general/test_time_velocity: 7.0
 * /hummingbird/agile_autonomy/general/velocity_estimate_in_world_frame: False
 * /hummingbird/agile_autonomy/input_cost_exponential: 0.0
 * /hummingbird/agile_autonomy/maneuver/length_straight: 40.0
 * /hummingbird/agile_autonomy/maneuver/maneuver_velocity: 7.0
 * /hummingbird/agile_autonomy/max_bodyrate_xy: 6.0
 * /hummingbird/agile_autonomy/max_bodyrate_z: 2.0
 * /hummingbird/agile_autonomy/max_thrust: 20.0
 * /hummingbird/agile_autonomy/min_thrust: 5.0
 * /hummingbird/agile_autonomy/p_B_C: [0.0, 0.0, 0.0]
 * /hummingbird/agile_autonomy/print_info: False
[... 999 lines cut ...]
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
Ready ? : 0
```

closed 2022-05-23 by tmddn833

boyuandeng, 2024-03-04:
I faced same error, the flightmare window is too slow(laggy, like frame loss). I tried your method but not work. Did you fixed it?

kcx1207, 2025-04-09:
mkdir cv_bridge_ws
cd cv_bridge_ws
mkdir src
cd src
git clone https://github.com/ros-perception/vision_opencv.git
// Find version of cv_bridge in your repository
cd vision_opencv
apt-cache show ros-noetic-cv-bridge | grep Version
	// 输出 Version: 1.16.2-1focal.20240913.193012
git checkout 1.16.2（替换为自己的版本1.16.2）
cd ../../
// Build
catkin build cv_bridge
// Extend environment with new package
source devel/setup.bash --extend

原文链接：https://blog.csdn.net/qq_44322965/article/details/131154293