uzh-rpg/agile_autonomy #53: Unable to run the Expert Planner

yanrui89, 2022-05-19:
I am trying to generate my own data and tried running the following command `python dagger_training.py --settings_file=config/dagger_settings.yaml` as suggested in the README. 

If I understand correctly, ln 102 of `dagger_training.py` calls the run_mppi_expert function, which is in ln 60 of this code `https://github.com/uzh-rpg/agile_autonomy/blob/main/planner_learning/src/PlannerLearning/PlannerLearning.py`. The run_mppi_expert function seems to be publishing to a topic called `/hummingbird/start_label`.

And this seems to be the place where the code gets stuck. and it seems to be just waiting for the expert to be done. However, I did a search on this repo and no where else seems to be subscribing to the `/hummingbird/start_label` topic. Hence, the code gets stuck. Anyone has any idea if I needed to run something else to get the expert running?

Thank you.

antonilo, 2022-05-20:
Can you try to increase[ the sleep time](https://github.com/uzh-rpg/agile_autonomy/blob/main/data_generation/agile_autonomy/src/flightmare_bridge.cpp#L114) before starting Unity?

mw9385, 2022-05-25:
Hi, I have a similar problem. After running the following command `python dagger_training.py --settings_file=config/dagger_settings.yaml`, the drone stands still and does not move even though enough time has passed. It kept repeating `reset` environment. 

When I debugged the python script `dagger_training,py`, I found that __self.learner.maneuver_complete__ always shows __False__ and I have difficulty in finding functions that updates the variable. Anyone has similar problems and solved the issue?

Thank you.

yanrui89, 2022-05-27:
> Can you try to increase[ the sleep time](https://github.com/uzh-rpg/agile_autonomy/blob/main/data_generation/agile_autonomy/src/flightmare_bridge.cpp#L114) before starting Unity?
[image]

Hi Antonio, thanks for your reply. However, it is still not working. So when I start running the dagger_training.py. I managed to get the above. So the quadcopter started running. Is this the expert running?

So the above was for the first experiment, subsequently, there was no progress to the second experiment and the process died, as seen in the image below.
[image]

yanrui89, 2022-05-27:
> Hi, I have a similar problem. After running the following command `python dagger_training.py --settings_file=config/dagger_settings.yaml`, the drone stands still and does not move even though enough time has passed. It kept repeating `reset` environment. By following your command on #17, I made the directory `data` in `data_generation` directory. Unfortunately it still doesn't work.
> 
> When I debugged the python script `dagger_training,py`, I found that **self.learner.maneuver_complete** always shows **False** and I have difficulty in finding functions that updates the variable. Anyone has similar problems and solved the issue?
> 
> Thank you.

May I know which part of the code you are looking at?

mw9385, 2022-05-27:
Hi, I think our code get stuck at the same line. My code stops at here.
https://github.com/uzh-rpg/agile_autonomy/blob/d95acdd8191c73aad4bb6f5bd454ff7a7ad76384/planner_learning/dagger_training.py#L79 When I check the code, the variable `self.learner.maneuver_complete` is update in callback function __callback_fly__ in `PlaneBase.py`. 
https://github.com/uzh-rpg/agile_autonomy/blob/d95acdd8191c73aad4bb6f5bd454ff7a7ad76384/planner_learning/src/PlannerLearning/PlannerBase.py#L77
The callback function is updated through the topic name `/hummingbird/agile_autonomy/start_flying`. So, I echoed the topic name `/hummingbird/agile_autonomy/start_flying`, but I didn't get any message.

So, I tried with some other topics that updates `/hummingbird/agile_autonomy/start_flying`. It is updated from the node `/test_primitive`. When I tried to echo some topics starting with `/test_primitive` (I cannot correctly remember this), I got some error message like:
`Error: Cannot load message class for [planning_ros_msgs/Trajectory]. Are your message built?`

Finally, I found that the topics from __global_planner__ is not correctly build. The package __rpg_mpl_ros__ has been ignored during the `catkin build` step. 

So, I think you should check whether your `rpg_mpl_ros` packages are correctly built. In my setting environment, when I tried to build the package isolated through the command `catkin build rpg_mpl_ros`, I got error message that the given package is not in the __src__ directory.

yanrui89, 2022-05-27:
> @mw9385 I am assuming you managed to resolve your issue and could run the dagger_training.py? I checked and there were no such error messages. Thanks for the advice.

mw9385, 2022-05-27:
@yanrui89 Not yet. I am still struggling. My code get stuck here as the variable `self.learner.maneuver_complete` is not updated to __true__. https://github.com/uzh-rpg/agile_autonomy/blob/d95acdd8191c73aad4bb6f5bd454ff7a7ad76384/planner_learning/dagger_training.py#L79

I will check the subscriber and publisher of `/hummingbird/start_label`. Keep in touch :)

yanrui89, 2022-05-27:
So above you mentioned that the 'So, I tried with some other topics that updates /hummingbird/agile_autonomy/start_flying. It is updated from the node /test_primitive.'

So the callback_fly is the callback function from the subscriber to the topic `/hummingbird/agile_autonomy/start_flying` and the publisher to the topic is this? 
https://github.com/uzh-rpg/agile_autonomy/blob/c9ded9a6ecf0aded8319850c81df23a65bf97960/data_generation/agile_autonomy/launch/simulation.launch#L59

Which seems like it is launching this node.
https://github.com/uzh-rpg/agile_autonomy/blob/c9ded9a6ecf0aded8319850c81df23a65bf97960/data_generation/agile_autonomy/src/agile_autonomy.cpp#L58

Is this similar to what you are thinking? I am not sure why the publisher to the /hummingbird/agile_autonomy/start_flying didn't get updated in your case...

mw9385, 2022-05-27:
@yanrui89 Hi, the topic `/hummingbird/agile_autonomy/start_flying` is initiated when the global planner finished making trajectories; however it seems that my global planner is not working. When I tried to reach the package through `roscd rpg_mpl_ros`, I cannot access to it although other packages such as agile_autonomy, flightrender are fine. Could you check whether you could reach the `rpg_mpl_ros` package through `roscd` command?

yanrui89, 2022-05-27:
@mw9385 I can't seem to access rpg_mpl_ros through the roscd command. But I was pretty sure there weren't any errors during the building of rpg_mpl_ros

mw9385, 2022-05-27:
@yanrui89 Thanks. I don't know what is the cause of problem then. I rebuild the given packages couple of times but the drone still doesn't move. Can docker environment cause the issue? I am build the packages in docker environment. 

So to sum up, you succeed in building without any errors. But you get stuck at some point and the drone cannot move. As your global planner can generate global trajectories (as we could see from your figures above), we need to check on  `generate_label` topic. Did you check the topics related to global planner (/test_primitive/~) are publishing messages?

antonilo, 2022-06-29:
Sorry but we do not provide support for docker

BAYMAXer, 2023-11-17:
I have the same problem. Did you solve it? thank you very much!

> > Can you try to increase[ the sleep time](https://github.com/uzh-rpg/agile_autonomy/blob/main/data_generation/agile_autonomy/src/flightmare_bridge.cpp#L114) before starting Unity?
> > [image]
> 
> Hi Antonio, thanks for your reply. However, I tried lengthening the sleep time but it is still not working.
> 
> So when I start running the dagger_training.py. I managed to get the above. So the quadcopter started running. Is this the expert running? It doesn't seem to be the expert running as it didn't go past
> 
> https://github.com/uzh-rpg/agile_autonomy/blob/d95acdd8191c73aad4bb6f5bd454ff7a7ad76384/planner_learning/dagger_training.py#L102
> 
> in the code.
> So my code is stuck right here at ln 102.
> 
> [image]

Zedzou, 2024-06-06:
hello, have you solve it? I have the same problem, but dont know how to solve.