uzh-rpg/agile_autonomy #54: Cannot see flightmare simulator after running simulation.lauch

mw9385, 2022-05-20:
Hi, I build the package by handing several issues and ran the code through the command `roslaunch agile_autonomy simulation.launch`. However, I could only see popup windows of rivz and controller GUI, which excludes the __flightmare simulator__. Also, it continuously shows the `Ready ?: 0 `. Are there any related issues or solutions for this? 
- hardware setup
- Docker: Ubuntu 20.04
- GPU: Titan X

I have changed the CMakelist in `agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt`, which is already handled in another issue [here](https://github.com/uzh-rpg/agile_autonomy/issues/18#issue-1068392852). I have refered this [blog](https://arnon.dk/matching-sm-architectures-arch-and-gencode-for-various-nvidia-cards/). 

One thing I've noticed from the error log is that the flightmare simulator in the `flightrender` is shutdown automatically. Following is the error log. 

```
[flight_render-20] process has died [pid 188249, exit code -11, cmd /root/agile_autonomy_ws/catkin_aa/src/rpg_flightmare/flightrender/standalone/flightmare.x86_64 __name:=flight_render __log:=/root/.ros/log/a514f09a-d81c-11ec-bece-88366cf6d71f/flight_render-20.log].
log file: /root/.ros/log/a514f09a-d81c-11ec-bece-88366cf6d71f/flight_render-20*.log
```

mw9385, 2022-05-24:
I solved it by reinstall cuda (version: 11.2) and cudnn. Also, add cuda path in bashrc.

closed 2022-05-24 by mw9385

stillstream, 2022-12-01:
I encounter the same problem and try to reinstall cuda11.2 , can you share your docker to me? Thank you very much!