uzh-rpg/agile_autonomy #56: About MPC implementation

coffear, 2022-06-16:
I would like to know if you use MPC in pure simulation, or only in real-world experiments?

antonilo, 2022-06-29:
MPC is our low-level controller, which we always use

coffear, 2022-07-29:
> MPC is our low-level controller, which we always use

I have been trying to replicate the work in simulation and real-world. However, neither in simulation nor real-world cases, neither low speed nor high speed, the vehicle is prone to crash into obstacles. Especially in the real world, the predicted trajectories are waving left and right when facing an obstacle, this results in the vehicle finally hitting on the obstacle. I feel really upset after pay months' of time on this project

antonilo, 2022-07-29:
I am very sorry that you are having trouble reproducing the work in the real world. Happy to help! Without knowing the details of your real-world system is difficult to debug, but I can try to guess.
The fact that the predictions are jittering in the real world indicates that your real-world controller is not fast enough to track the predicted trajectory. Having a fast enough controller can help to solve the jittering problem. To improve temporal consistency, consider decreasing the acceptance threshold or adding temporal consistency in the training loss.