uzh-rpg/agile_autonomy #60: The predicted trajectories are switching between left and right when flying towards an obstacle

coffear, 2022-07-29:
Both in simulation and real-world experiments, I find that the predicted trajectories will be switching between left and right when the vehicle is flying towards an obstacle.  For the neural network, it is natural because it is just trying to avoid the obstacle. The problem is that it keeps this kind of switch, finally leading the vehicle to crash on the obstacle.  In another word, when there is an obstacle in front of the quadrotor, the neural network tells the robot to go to the left to avoid the obstacle, so the quadrotor moves to the left, maybe after a second, the neural network asks the robot to go to the right to avoid the obstacle.... then again left, right, left, right......

In your paper and the codes, the processes of fitting by polynomials or projecting the trajectory obtained from the raw predicted trajectory are basically trying to minimize the control input cost. But I do not think this can systematically solve the problem mentioned above. 

However, making the decision on which side to go to avoid the obstacle is quite important for an "AI" system. I do not see any clear explanation in the paper, and also failed to do it in the real-world experiments

antonilo, 2022-07-29:
Hi! Thanks for your detailed feedback! The switching between modes could be a problem. One way to control it is decreasing [the acceptance threshold](https://github.com/uzh-rpg/agile_autonomy/blob/main/planner_learning/config/deploy_settings.yaml#L37). In this case more trajectories will pass to the fitting, and the predictions will be more consistent. We make the decision based on energy minimization (we discuss this in the paper). Minimizing energy will bias trajectory to be temporally consistent.
But you can always construct some cases where it will fail. Some [recent work](https://www.researchgate.net/publication/362075736_Deep-PANTHER_Learning-Based_Perception-Aware_Trajectory_Planner_in_Dynamic_Environments) proposed new losses to ease this problem.

coffear, 2022-08-01:
@antonilo BTW, do you have a low-speed trained model ( < 2m/s on average ), if you have it, could you share it with me, it could save me a lot of time to verify the problem. My personal PC is not good, and it is not smooth when I am controlling the flight in obtaining the training data

LeiOuYang, 2023-04-19:
@coffear I also want to reproduce this project recently, but I don't know how to build a real environment. The project seems to only mention how to simulate and train. Do you have more detailed instructions on how to build and use the real machine? Thanks

antonilo, 2023-04-19:
You can get instructions on how to build and control the drone here: https://github.com/uzh-rpg/agilicious