uzh-rpg/agile_autonomy #66: Build is failing for Open3D

saurabhprasun20, 2022-10-01:
Hello,

I am trying to build the project but its failing while running the build for open3d_conversion package. I am getting below error --

```
 fatal error: Open3D/Open3D.h: No such file or directory
 #include <Open3D/Open3D.h>

```
I have build Open3D from source using the instruction in this link --> http://www.open3d.org/docs/release/compilation.html.
I was able to build and verify the build by using the instruction in this repo --> https://github.com/isl-org/open3d-cmake-find-package

So Open3D is working in my system.

I also checked the CMakeLists.txt file in package src/rpg_mpl_ros/open3d_conversions and I found that its trying to find the open3d by using this statement --

 `find_package(Open3D HINTS /usr/local/lib/cmake/)`

I also checked my /usr/local/lib/cmake library and it has a open3d folder present in it with below files inside  -->

```
Open3DConfig.cmake         Open3DTargets.cmake
Open3DConfigVersion.cmake  Open3DTargets-release.cmake
```

I am a beginner with c++ and CMake and so I don't know if it's trying to find open3d.h file here.

Can someone please let me know what might have gone wrong with the build?

coffear, 2022-10-02:
install opend3d v0.9.0

saurabhprasun20, 2022-10-04:
> 

Thanks!

Installing Open3d v0.9.0 worked for me although it was a bit tricky.

But now I am getting a new error as below for the NUMPY-EIGEN module-->\

```
In file included from /usr/include/x86_64-linux-gnu/bits/types/stack_t.h:23:0,
                 from /usr/include/signal.h:303,
                 from /home/prasun/anaconda3/lib/python3.9/site-packages/numpy/core/include/numpy/npy_interrupt.h:23,
                 from /home/prasun/anaconda3/lib/python3.9/site-packages/numpy/core/include/numpy/arrayobject.h:5,
                 from /home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/include/numpy_eigen/NumpyEigenConverter.hpp:28,
                 from /home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp:2:
/home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp: In function ‘void init_module_libnumpy_eigen()’:
**/home/prasun/anaconda3/lib/python3.9/site-packages/numpy/core/include/numpy/__multiarray_api.h:1532:151: error: return-statement with a value, in function returning 'void' [-fpermissive]
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NULL; } }**
```

I am also getting one warning which I don't know if I should resolve. The warning is as below -->

```
In file included from /home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/include/numpy_eigen/NumpyEigenConverter.hpp:30:0,
                 from /home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp:2:
/home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/include/numpy_eigen/type_traits.hpp: In static member function ‘static bool TypeToNumPy<unsigned char>::canConvert(int)’:
/home/prasun/agile_autonomy_ws/catkin_aa/src/numpy_eigen/include/numpy_eigen/type_traits.hpp:63:61: warning: ‘NPY_CHAR’ is deprecated: Use NPY_STRING [-Wdeprecated-declarations]
     return type == NPY_UBYTE || type == NPY_BYTE || type == NPY_CHAR;
```

Can you please suggest the solution for the error and should I skip the warning or change the data type from NPY_CHAR to NPY_STR?

closed 2022-10-04 by saurabhprasun20

duy12i1i7, 2024-02-11:
@saurabhprasun20 
<div class="x_elementToProof"><span style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)">Dear Prasun Saurabh.</span></div><div class="x_elementToProof"><span style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)">First of letter, Hope you have a good holyday about Lunar New Year.</span></div><div class="x_elementToProof"><span style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)"><br aria-hidden="true"></span></div><div class="x_elementToProof"><span style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)">I have a question for you about how to install 'uzh-rpg/agile_autonomy' from this link:<a href="https://github.com/uzh-rpg/agile_autonomy/" target="_blank" rel="noopener noreferrer" data-auth="NotApplicable" id="LPlnk" class="x_OWAAutoLink" data-linkindex="0">https://github.com/uzh-rpg/agile_autonomy/</a></span></div><div class="x_elementToProof"><span style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)"><br aria-hidden="true"></span></div><div class="x__Entity x__EType_OWALinkPreview x__EId_OWALinkPreview x__EReadonly_1"><div id="LPBorder_GTaHR0cHM6Ly9naXRodWIuY29tL3V6aC1ycGcvYWdpbGVfYXV0b25vbXkv" class="x_LPBorder582168" style="width: 100%; margin-top: 16px; margin-bottom: 16px; max-width: 800px; min-width: 424px; position: relative;">
  | GitHub  - uzh-rpg/agile_autonomy: Repository Containing the Code associated  with the Paper: "Learning High-Speed Flight in the Wild"Repository  Containing the Code associated with the Paper: &quot;Learning  High-Speed Flight in the Wild&quot; - GitHub -  uzh-rpg/agile_autonomy: Repository Containing the Code associated with  the Pa...github.com
-- | --

</div></div><div class="x_elementToProof" style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)">But
 I can not know how to install Open3D v0.9.0.0 like you, I tried alot of
 methods but cannot solve this problem, can you help me installing this 
library.</div><div class="x_elementToProof" style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)"><br aria-hidden="true"></div><div class="x_elementToProof" style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)">Thanks for reading this letter. Have a good day!</div><div class="x_elementToProof" style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)"><br aria-hidden="true"></div><div class="x_elementToProof" style="font-family:Aptos,Aptos_EmbeddedFont,Aptos_MSFontService,Calibri,Helvetica,sans-serif; font-size:12pt; color:rgb(0,0,0)"><br aria-hidden="true"></div>Dear Prasun Saurabh.
First of letter, Hope you have a good holyday about Lunar New Year.

I have a question for you about how to install 'uzh-rpg/agile_autonomy' from this link:https://github.com/uzh-rpg/agile_autonomy/

	
[GitHub - uzh-rpg/agile_autonomy: Repository Containing the Code associated with the Paper: "Learning High-Speed Flight in the Wild"](https://github.com/uzh-rpg/agile_autonomy/)
Repository Containing the Code associated with the Paper: &quot;Learning High-Speed Flight in the Wild&quot; - GitHub - uzh-rpg/agile_autonomy: Repository Containing the Code associated with the Pa...
github.com


I have a problem about cannot include the Open3D library like that:

 fatal error: Open3D/Open3D.h: No such file or directory
 #include <Open3D/Open3D.h>

When I found the solution to fix this, I saw your issue on Github(https://github.com/uzh-rpg/agile_autonomy/issues/66)
	
[Build is failing for Open3D · Issue #66 · uzh-rpg/agile_autonomy](https://github.com/uzh-rpg/agile_autonomy/issues/66)
Hello, I am trying to build the project but its failing while running the build for open3d_conversion package. I am getting below error -- fatal error: Open3D/Open3D.h: No such file or directory #i...
github.com
But I can not know how to install Open3D v0.9.0.0 like you, I tried alot of methods but cannot solve this problem, can you help me installing this library.

Thanks for reading this letter. Have a good day!