uzh-rpg/agile_autonomy #75: Inconsistent references when restoring backbone checkpoint data

RaffaeleBrilli, 2023-02-17:
Hi! First of all, thanks for publishing this great work!
I'm having some troubles in running the network using the test_trajectories.py script with the ckpt file you provided.
It seems that it is not able to properly restore the tensors of the depth backbone from the ckpt.
I'm posting below the full set of warnings that tensorflow is printing.
Are you aware of a solution to this issue?
Thanks a lot.

WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.Conv2D object at 0x0000028983459DC8> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898346A848>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.Conv2D object at 0x0000028983459DC8> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898346A848>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898346A848> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898F968088>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898346A848> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898F968088>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.DepthwiseConv2D object at 0x000002898FB54988> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB59108>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.DepthwiseConv2D object at 0x000002898FB54988> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB59108>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB59108> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898FB65D08>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB59108> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898FB65D08>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.Conv2D object at 0x000002898FB6B908> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB6EC48>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.Conv2D object at 0x000002898FB6B908> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB6EC48>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB6EC48> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898FB78808>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB6EC48> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898FB78808>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.DepthwiseConv2D object at 0x000002898FB80F48> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB864C8>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.convolutional.DepthwiseConv2D object at 0x000002898FB80F48> and <tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB864C8>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB864C8> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898FB8D8C8>).
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002898FB864C8> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002898FB8D8C8>).
[... cut ...]
WARNING:tensorflow:Inconsistent references when loading the checkpoint into this object graph. Either the Trackable object references in the Python program have changed in an incompatible way, or the checkpoint was generated in an incompatible program.

Two checkpoint references resolved to different objects (<tensorflow.python.keras.layers.normalization_v2.BatchNormalization object at 0x000002899A12DF08> and <tensorflow.python.keras.layers.advanced_activations.ReLU object at 0x000002899B4DD688>).

blu666, 2023-03-09:
Hi, I've run into the same problem. Have you found a solution yet?

Liviy-yu, 2023-03-29:
me 2 T.T Have you solve it?

joaobose, 2023-08-25:
I'm facing the same Issue, does anyone have a solution?

phoenixrider12, 2025-06-18:
Was anyone able to solve this?