uzh-rpg/agile_autonomy #78: How to Create a Custom Simulation Environment？

xiaolaji0426, 2023-04-11:
I have known that we  should render the environment by unity.

And in the file `rosed agile_autonomy flightmare.yaml` we can change the `env_idx` to set different pre-set envs.

But I wonder where are these pre-env(streets. wasteland, etc) files located? I have found out the whole  folder `rpg_flightmare` and have no idea.

Additionally, how can I create other env beside those mentioned in the paper?

Thanks a lot !

antonilo, 2023-04-16:
You can download all environments from the unity asset store https://assetstore.unity.com/

maplecod, 2024-01-23:
Do you know how to add a new environment?

mw9385, 2024-08-07:
@maplecod Hi, did you solve the problem?

maplecod, 2024-08-07:
@mw9385 You can't add a new environment, the forest density can be modified

mw9385, 2024-08-07:
@maplecod Thanks. I need to stick to the original environments.