uzh-rpg/agile_autonomy #81: run roslaunch agile_autonomy simulation.launch bug log：no kernel image is available for execution on the device 209

PDLwork, 2023-06-13:
## Issue Description
When I run roslaunch agile_autonomy simulation.launch
[ERROR] [1686655639.396501363, 3.536000000]: no kernel image is available for execution on the device 209

The details are as follows:
Ready?: 0
Ready?: 1
[INFO] [1686655639.023764265, 3.349000000]: Unity is ready!
[ERROR] [1686655639.396501363, 3.536000000]: no kernel image is available for execution on the device 209
[ERROR] [1686655639.531226480, 3.603000000]: no kernel image is available for execution on the device 209

## Environment Information
- Operating System: Ubuntu 20.04
- GPU Version: RTX3060
- CUDA Version： 12.1 and 11.0  (The above error occurs)

## Steps to Reproduce
I finally catkin build successfully according to readme and other blogs, but when running roslaunch agile_autonomy simulation.launch, three windows can be successfully opened, but when the picture in the simulation is displayed in rviz, the error mentioned above occurs.

Thank you！

BAYMAXer, 2023-10-24:
I meet the same situation. Do you solve this problem? Thanks

maplecod, 2023-12-19:
Do you solve this problem?

mw9385, 2024-08-06:
mainly cause by the GPU architecture. Search your GPU architecture from NVIDIA and then add them in CMakelist

fjlv520, 2025-01-16:
in src/agile_autonomy/data_generation/agile_autonomy/CMakeLists.txt
add line 20 with 
-gencode=arch=compute_86,code=sm_86
and  catkin build again
then in ~/.bashrc file add with 
TORCH_CUDA_ARCH_LIST="8.0"