uzh-rpg/agile_autonomy #83: Run python train.py --settings_file=config/train_settings.yaml Error: Segment error (core has been dumped)

Nuyoah0123, 2023-06-15:
```javascript
2023-06-15 22:06:58.512423: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
2023-06-15 22:07:00.958351: I tensorflow/compiler/jit/xla_cpu_device.cc:41] Not creating XLA devices, tf_xla_enable_xla_devices not set
2023-06-15 22:07:00.958428: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcuda.so.1
2023-06-15 22:07:00.958508: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:941] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2023-06-15 22:07:00.958592: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1720] Found device 0 with properties: 
pciBusID: 0000:01:00.0 name: NVIDIA GeForce RTX 3090 Ti computeCapability: 8.6
coreClock: 1.86GHz coreCount: 84 deviceMemorySize: 23.65GiB deviceMemoryBandwidth: 938.86GiB/s
2023-06-15 22:07:00.958600: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
2023-06-15 22:07:00.960492: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcublas.so.11
2023-06-15 22:07:00.960567: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcublasLt.so.11
2023-06-15 22:07:00.961168: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcufft.so.10
2023-06-15 22:07:00.961307: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcurand.so.10
2023-06-15 22:07:00.961379: W tensorflow/stream_executor/platform/default/dso_loader.cc:60] Could not load dynamic library 'libcusolver.so.10'; dlerror: libcusolver.so.10: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /home/llj/anaconda3/envs/tf_24/lib/python3.7/site-packages/cv2/../../lib64:/home/llj/agile_autonomy_ws/catkin_aa/devel/lib:/opt/ros/noetic/lib:/usr/local/cuda/lib64:/usr/local/cuda/lib64
2023-06-15 22:07:00.961731: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcusparse.so.11
2023-06-15 22:07:00.961805: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudnn.so.8
2023-06-15 22:07:00.961810: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1757] Cannot dlopen some GPU libraries. Please make sure the missing libraries mentioned above are installed properly if you would like to use GPU. Follow the guide at https://www.tensorflow.org/install/gpu for how to download and setup the required libraries for your platform.
Skipping registering GPU devices...
2023-06-15 22:07:00.962995: I tensorflow/compiler/jit/xla_gpu_device.cc:99] Not creating XLA devices, tf_xla_enable_xla_devices not set
2023-06-15 22:07:00.963011: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1261] Device interconnect StreamExecutor with strength 1 edge matrix:
2023-06-15 22:07:00.963014: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1267]      
------------------------------------------
Initializing from scratch.
------------------------------------------
Training Network
Found 1245 images  belonging to 17 experiments:
Found 1245 images  belonging to 17 experiments:
/home/llj/anaconda3/envs/tf_24/lib/python3.7/site-packages/tensorflow/python/keras/backend.py:434: UserWarning: `tf.keras.backend.set_learning_phase` is deprecated and will be removed after 2020-10-11. To update it, simply pass a True/False value to the `training` argument of the `__call__` method of your layer or model.
  warnings.warn('`tf.keras.backend.set_learning_phase` is deprecated and '
  0%|                                 | 0/156 [00:00<?, ?it/s]2023-06-15 22:07:06.362969: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:116] None of the MLIR optimization passes are enabled (registered 2)
2023-06-15 22:07:06.381784: I tensorflow/core/platform/profile_utils/cpu_utils.cc:112] CPU Frequency: 2995200000 Hz
 48%|███████████▌            | 75/156 [00:11<00:08,  9.93it/s]段错误 (核心已转储)
```
What is this problem and how should I fix it ？

Nuyoah0123, 2023-06-15:
Will there be a problem if the version of open3d in the virtual environment tf_24 is not the same as the locally compiled version of Open_3D?

sdhar16, 2023-06-15:
Tensorflow cannot detect your GPU. Correct the LD_LIBRARY_PATH and try to run some examples on tensorflow to make sure that the gpu is being used, otherwise it will crash

Nuyoah0123, 2023-06-17:
@sdhar16 Thanks for your answer, I solved my problem with the following command
```javascaript
cd /usr/local/cuda-11.2/lib64 
ls -al 
mv libcusolver.so.11 libcusolver.so.10
```

closed 2023-06-17 by Nuyoah0123