uzh-rpg/agile_autonomy #84: After running python dagger_training.py --settings_file=config/dagger_settings.yaml an exception occurs and it keeps printing Crashing into something!

Nuyoah0123, 2023-06-23:
I tried to follow the tutorial in the ``Collect your own dataset`` section of ``Readme`` to collect my own data, I changed the ``resume_training`` parameter in the ``dagger_settings.yaml`` file to ``False`` and ran the ``dagger_training.py`` file, but it ran The result doesn't seem to be satisfactory.
[image]
As shown in the picture, the drone started after a short flight according to the reference trajectory, spinning in place and getting stuck in the middle of an obstacle.
I would like to know why this is happening and how I should fix the problem?

Nuyoah0123, 2023-06-23:
I plotted the ``xyz`` data from the ``ellipsoid_trajectory.csv`` file and the ``reference_trajectory.csv`` file, and the results are shown below
[image]

sdhar16, 2023-10-12:
Change the fallback radius to 0 during data collection when the network in initialised from scratch