uzh-rpg/agile_autonomy #86: rqt_gui version to run the repo? qt_gui_cpp plugin error

AnshShah3009, 2023-10-11:
```
RosPluginProvider.load(qt_gui_cpp/CppPluginProvider) exception raised in __builtin__.__import__(qt_gui_cpp.cpp_plugin_provider, [CppPluginProvider]):
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 79, in load
    module = __builtin__.__import__(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_plugin_provider.py", line 33, in <module>
    from .cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name

RecursivePluginProvider.discover() loading plugin "qt_gui_cpp/CppPluginProvider" failed:
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/recursive_plugin_provider.py", line 60, in discover
    instance = self._plugin_provider.load(plugin_descriptor.plugin_id(), None)
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 90, in load
    raise e
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 79, in load
    module = __builtin__.__import__(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_plugin_provider.py", line 33, in <module>
    from .cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name
```

What is the version of rqt that has been used in experimentation and how was the pyqt5 environment setup in the base environment, I am getting a failed process when running the simulation launch file:
```
roslaunch agile_autonomy simulation.launch
```

RibhavOjha, 2023-10-15:
I was getting problems when running `roslaunch agile_autonomy simulation.launch`, I deactivated both the  envs (base and the other one) and ran it and somehow it worked.

AnshShah3009, 2023-10-20:
@RibhavOjha  What is the opencv version you are using in all the environments

closed 2023-11-06 by AnshShah3009

ChanJoon, 2025-01-15:
@AnshShah3009 Hi did you resolve this issue?

AnshShah3009, 2025-01-15:
> @AnshShah3009 Hi did you resolve this issue?

I did resolve it, and it was related to library versions of opencv and qt but I can't remember exactly what changes I made. This is from too of my head, it could be that there were some other issues too.

ChanJoon, 2025-01-16:
> I did resolve it, and it was related to library versions of opencv and qt but I can't remember exactly what changes I made. This is from too of my head, it could be that there were some other issues [...]

Thank you. I resolved the issues and successfully launched the simulation.

The problem was related to Conda. The Conda environment should be deactivated in the terminal used to launch the simulation, but the PATH variable must still include Conda path. However, when building using `catkin build`, the Conda environment should also be deactivated, and the PATH variable must not include Conda path.

This mismatch caused several issues during the catkin build process and simulation runs.

As you mentioned, there were additional issues related to the rotors_simulator and a symbol lookup error, but they fall outside the scope of this explanation, so I will not address them here.

Anyway, thanks again for your kind response!!