uzh-rpg/agile_autonomy #87: Build failing when using catkin Build

RibhavOjha, 2023-10-15:
Hi
I am in the catkin build step. The build fails with most of them being abandoned and few failing. Can you please help.
```
 catkin build
-----------------------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa
-----------------------------------------------------------------------------------------------
Build Space:        [exists] /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/build
Devel Space:        [exists] /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/devel
Install Space:      [unused] /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/install
Log Space:          [exists] /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/logs
Source Space:       [exists] /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/src
DESTDIR:            [unused] None
-----------------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-fdiagnostics-color
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-----------------------------------------------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------------------------------------------
[build] Found 71 packages in 0.0 seconds.                                                                                                                                                                 
[build] Package table is up to date.                                                                                                                                                                      
Starting  >>> catkin_boost_python_buildtool                                                                                                                                                               
Starting  >>> catkin_simple                                                                                                                                                                               
Starting  >>> decomp_util                                                                                                                                                                                 
Starting  >>> mav_msgs                                                                                                                                                                                    
Starting  >>> mav_state_machine_msgs                                                                                                                                                                      
Starting  >>> mav_system_msgs                                                                                                                                                                             
Starting  >>> motion_primitive_library                                                                                                                                                                    
Starting  >>> open3d_conversions                                                                                                                                                                          
Finished  <<< catkin_simple                                     [ 0.1 seconds ]                                                                                                                           
Starting  >>> assimp_catkin                                                                                                                                                                               
Finished  <<< catkin_boost_python_buildtool                     [ 0.2 seconds ]                                                                                                                           
Finished  <<< mav_system_msgs                                   [ 0.5 seconds ]                                                                                                                           
Finished  <<< mav_state_machine_msgs                            [ 0.6 seconds ]                                                                                                                           
Starting  >>> catkin_boost_python_buildtool_test                                                                                                                                                          
__________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << open3d_conversions:make /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/logs/open3d_conversions/build.make.004.log                                                               
In file included from /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/src/open3d_conversions.cpp:15:0:
/home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/src/rpg_mpl_ros/open3d_conversions/include/open3d_conversions/open3d_conversions.h:19:10: fatal error: Open3D/Open3D.h: No such file or directory
 #include <Open3D/Open3D.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/open3d_conversions.dir/build.make:63: CMakeFiles/open3d_conversions.dir/src/open3d_conversions.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:305: CMakeFiles/open3d_conversions.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/ribhavojha/Desktop/agil2/agile_autonomy_ws/catkin_aa/build/open3d_conversions; catkin build --get-env open3d_conversions | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

..........................................................................................................................................................................................................
Failed     << open3d_conversions:make                           [ Exited with code 2 ]                                                                                                                    
Failed    <<< open3d_conversions                                [ 0.8 seconds ]                                                                                                                           
Abandoned <<< decomp_ros_msgs                                   [ Unrelated job failed ]                                                                                                                  
Abandoned <<< eigen_catkin                                      [ Unrelated job failed ]                                                                                                                  
Abandoned <<< flightrender                                      [ Unrelated job failed ]                                                                                                                  
Abandoned <<< gflags_catkin                                     [ Unrelated job failed ]                                                                                                                  
Abandoned <<< minimum_jerk_trajectories                         [ Unrelated job failed ]                                                                                                                  
Abandoned <<< planning_ros_msgs                                 [ Unrelated job failed ]                                                                                                                  
Abandoned <<< quad_launch_files                                 [ Unrelated job failed ]                                                                                                                  
[... cut ...]
[build]   Warnings:  None.                                                                                                                                                                                
[build]   Abandoned: 61 packages were abandoned.                                                                                                                                                          
[build]   Failed:    2 packages failed.                                                                                                                                                                   
[build] Runtime: 2.5 seconds total.   
```

Arcane-01, 2024-01-16:
The issue for open3d_conversions can be resolved by building Open3D from source and renaming the include file of open3d to `#include "open3d/Open3D.h"`. You can refer to #10, it addresses a similar problem.

yifeichen2024, 2024-07-16:
I have the same problem with the second package, did you solve it?

RibhavOjha, 2024-07-16:
> I have the same problem with the second package, did you solve it?

I stopped using agile_autonomy and shifted to agilicious instead which is a superior and better flight stack and allows easier integration  into your personal projects.

yifeichen2024, 2024-07-18:
Ok. I just solve this problem, I will try playing with it a little bit. I will also try agilicious. Thanks