uzh-rpg/agile_autonomy #88: Bad Callback on running test script

AnshShah3009, 2023-10-17:
[ERROR] [1697558138.151880, 148.512000]: bad callback: <bound method PlanBase.callback_depth of <PlannerLearning.PlannerLearning.PlanLearning object at 0x7f2a3e48b450>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/baymax/Desktop/agile_autonomy_ws/catkin_aa/src/agile_autonomy/planner_learning/src/PlannerLearning/PlannerBase.py", line 206, in callback_depth
    depth = self.bridge.imgmsg_to_cv2(data, '16UC1')
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 163, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: /lib/x86_64-linux-gnu/libp11-kit.so.0: undefined symbol: ffi_type_pointer, version LIBFFI_BASE_7.0

AnshShah3009, 2023-10-17:
From my analysis, it looks like an opencv and rospy compatiblity issue, Can someone suggest what version should i use

Hofalen, 2023-12-14:
i have the same issue, did you fix it?

AnshShah3009, 2023-12-14:
@Hofalen The problem was related to libffi version in the conda environment. Try installing earlier subversion(3.7.x) of python in the conda env, should solve the issue.

Hofalen, 2023-12-14:
@AnshShah3009 thanks for the quick answer! I proceed to install python 3.7.8 and it worked

Cryptheon, 2024-01-01:
Another option would be to use ros_numpy and replace the lines of code with an equivalent (use numpify from ros) you don't need cv_bridge in this case.

ChanJoon, 2025-09-24:
I solved the issue easily. Thanks to @Cryptheon . For the reference:

```diff
-from cv_bridge import CvBridge, CvBridgeError
+import ros_numpy
 from nav_msgs.msg import Odometry
 from rpg_quadrotor_msgs.msg import TrajectoryPoint
 from rpg_quadrotor_msgs.msg import Trajectory
@@ -37,7 +36,6 @@ class PlanBase(object):
         self.last_depth_received = rospy.Time.now()
         self.reference_progress = 0
         self.reference_len = 1
-        self.bridge = CvBridge()
         self.mode = mode
         self.image = np.zeros(
             (self.config.img_height, self.config.img_width, 3))
@@ -190,12 +188,16 @@ class PlanBase(object):
         Reads an image and generates a new plan.
         '''
         try:
-            image = self.bridge.imgmsg_to_cv2(data, "bgr8")
+            image = ros_numpy.numpify(data)
+            if data.encoding == 'rgb8':
+                image = cv2.cvtColor(image, cv2.COLOR_RGB2BGR)
+            elif data.encoding == 'mono8':
+                image = cv2.cvtColor(image, cv2.COLOR_GRAY2BGR)
             if self.quad_name == 'hawk':
                 image = cv2.flip(image, -1)  # passing a negative axis index flips both axes
             if np.sum(image) != 0:
                 self.image = self.preprocess_img(image)
-        except CvBridgeError as e:
+        except Exception as e:
             print(e)
 
     def preprocess_depth(self, depth):
@@ -213,28 +215,21 @@ class PlanBase(object):
         Reads a depth image and saves it.
         '''
         try:
-            if self.quad_name == 'hummingbird':
-                depth = self.bridge.imgmsg_to_cv2(data, '16UC1')
-                # print("============================================================")
-                # print("Min Depth {}. Max Depth {}. with Nans {}".format(np.min(depth),
-                #                                                        np.max(depth),
-                #                                                        np.any(np.isnan(depth))))
-                # print("Min Depth {}. Max Depth {}. with Nans {}".format(np.min(depth),
-                #                                                        np.max(depth),
-                #                                                        np.any(np.isnan(depth))))
-            elif self.quad_name == 'hawk':
+            depth = ros_numpy.numpify(data)
+            if depth.ndim == 3:
+                depth = depth[:, :, 0]
+            if self.quad_name == 'hawk':
                 # print("Received depth image from hawk!")
```