uzh-rpg/agile_autonomy #89: Catkin build rotors_gazebo_plugins Failed

GavindotYang, 2023-12-13:
Hi,there are some problem that I hope to find some help from you.
I am doing catkin build but : I meet many errors saying "error: no matching function for call to....."

[image]

Ulricall, 2024-01-15:
Hi, have you solved this issue? I encounter the same problem. I've tried to add "set(CMAKE_CXX_STANDARD 17)" at the beginning of the cmakelists.txt but it doesn't work.

Arcane-01, 2024-05-30:
Hi @Ulricall 
Did you manage to resolve this issue? Previously I had resolved it by adding set(CMAKE_CXX_STANDARD 17) to the CMakelists.txt
But on a new system, even after adding this line, the issue is not resolved.