uzh-rpg/agile_autonomy #91: train neural  network

w2385680516, 2024-01-18:
Hello author, what trick did you use when training the neural network? In the forest scene, I used the train.py file directly, but the trained model could never achieve the same effect as your model. can you help me? Thanks.

antonilo, 2024-03-18:
We trained on multiple scenes and a large number of environments. The quickest way to do this is to download the data from the associated link and train on our previously collected dataset. After a finetuning round at the speed you need to go, the network would reach approximately (accounting for the randomness in the simulator) the same performance of our checkpoint.

mmz-mmz-mmz123, 2025-04-20:
what trick did you use when training the neural network? In the forest scene, I used the train.py file directly, but the trained model could never achieve the same effect as your model.